hannobraun/fornjot #2: Feature: Support for GD&T

Bandsberg, 2021-11-08:
Would you be interested in supporting [GD&T](https://en.wikipedia.org/wiki/Geometric_dimensioning_and_tolerancing)?

I would love to help implementing this if it has interest. I am not an expert, but we use it (ISO GPS to be specific) in the company that I work at.

hannobraun, 2021-11-08:
Hey @Bandsberg, thank you for your interest and for opening this issue! Yes, I'd be interested in supporting GD&T, in theory. In practice, I think we're not nearly at the point were it makes sense to seriously consider.

To explain the context, the code in the top-level directory (the one I've been working on for a while) is probably a dead end. I've just started working on a new prototype that's basically going to be a rewrite (with lots of reused code, mainly on the rendering side). It's available in the [`ng/`](https://github.com/hannobraun/fornjot/tree/main/ng) directory. Check out the README there, for more information.

So while I'd love to have support for GD&T (I *just* [did some really bad drawings in Inkscape](https://github.com/hannobraun/prusa-mini-enclosure), and it sucked :grin:), I fear that right now the technical base it would need to be built on is severely lacking.

I also think that the project hasn't reached a level of maturity yet where it could really support contributors, as long as I'm going around, creating new prototypes, deleting old code, and such. And don't get me wrong, I'd **love** to have help around here, and if you're interested in that, we should talk some more! But what I'm saying is, even if you submitted the perfect GD&T implementation today, I couldn't guarantee that it would still be present in the latest actively developed version a month from now.

So yeah, if you want to get involved under these circumstance, I'd love to talk about that. I'm just trying to set expectations properly. This project is _really_ experimental, to the point where most people probably wouldn't have published it yet (I, on the other hand, know no shame when it comes to open source :smile:).

Bandsberg, 2021-11-09:
@hannobraun Thank you for the good elaboration of your answer and aligning expectations! ☺️ 
You make some very good points.
I think for now I will follow the progress of the code and see when the time is right to contribute ☺️ 

P.S. Love that you are not shameful 😄

closed 2021-11-09 by Bandsberg

hannobraun, 2021-12-28:
Since my last comment in this issue, a lot of things have changed. Since that was a broad overview of the state of the project, and parts of it are no longer true, I'd like to post an update here.

First, the prototype I was talking about has replaced the (former) main code. It's made good progress since then, and while I would still consider Fornjot to be experimental, I can see a much clearer path forward now than ever before.

Second, my statement that the project can't really support contributors is no longer true. Things have stabilized significantly since then, and I've made an effort to open actionable issues that others can pick up. (And in fact, @Bandsberg, who opened this issue, has started contributing since then.)

Third, I still think it's too early to seriously consider GD&T support. A lot of the code that such a feature would have to be built upon is still in flux, so it would either get in the way, or risk being removed again. However, this is starting to change slowly. The API that is growing in the [`kernel`](https://github.com/hannobraun/fornjot/tree/6da79278480e3a8b805577af3d101d8b96bc067d/src/kernel) module, most relevantly in places like [`edges`](https://github.com/hannobraun/fornjot/blob/6da79278480e3a8b805577af3d101d8b96bc067d/src/kernel/edges.rs) and [`faces`](https://github.com/hannobraun/fornjot/blob/6da79278480e3a8b805577af3d101d8b96bc067d/src/kernel/faces.rs) is slowly evolving, and I wouldn't be surprised if my view on GD&T support were to change in a few months.

hannobraun, 2022-02-08:
I've added this to the new [Feature Wishlist](https://github.com/hannobraun/Fornjot/discussions/146).