hannobraun/fornjot #11: Make sure model libraries are handled correctly on all platforms

hannobraun, 2021-11-22:
Right now, the host application expects the dynamic library to end with `.so`, which is correct on Linux. It should either fall back to other file endings to support more platforms, or there might be a way to override the file name outright, and set it to something like `.model`.

hannobraun, 2021-11-22:
In case anyone wants to take a look, the library is loaded in [`model.rs`](https://github.com/hannobraun/fornjot/blob/06d90d846c38249466409a73e3a398f3d46de49f/src/model.rs#L43). If it's practical to override the whole file name, this [also needs to happen in `model.rs`](https://github.com/hannobraun/fornjot/blob/06d90d846c38249466409a73e3a398f3d46de49f/src/model.rs#L30-L33)

hannobraun, 2021-11-23:
I initially thought of this as an enhancement (as in, it works fine, it should work fine in more places), but I think it makes more sense to classify it as a bug.

Bandsberg, 2021-12-15:
For OSX: dynamic library: .dylib (e.g. libspacer.dylib) results in "Error: Error request adapter". Error defined in renderer.rs
Will continue to investigate next time I have time

hannobraun, 2021-12-15:
@Bandsberg I think what you're seeing is most likely #10, not this issue.

hannobraun, 2021-12-17:
Fixed by @Bandsberg in #23.

closed 2021-12-17 by hannobraun