hannobraun/fornjot #13: Display model structure

hannobraun, 2021-11-23:
Right now, only the end result is shown. The primitives and operations applied to them, that make up the model, aren't. This can make it hard to debug your model.

There should be some kind of tree structure visible on the screen, that shows all those primitives and operations which make up the model. Each element in that tree should be selectable. When selected, only that element should be shown.

Blocked on #4.

hannobraun, 2022-03-04:
Relevant thoughts: https://github.com/hannobraun/Fornjot/issues/220#issuecomment-1059144575

hannobraun, 2022-07-11:
No longer blocked. As of #763, we have support for egui now.

Michael-F-Bryan, 2022-07-11:
> There should be some kind of tree structure visible on the screen, that shows all those primitives and operations which make up the model. Each element in that tree should be selectable. When selected, only that element should be shown.

This sounds really similar to the system used by SolidWorks when doing mechanical design.

The idea is you have some sort of "feature tree" down the left hand side where one feature might be a 2D sketch, then the next feature is an extrude of that sketch, and the third feature might be adding fillets to the extruded shape. 

As well as being able to right-click and "suppress" a feature and any features that depend on it (e.g. suppressing the extrude will also hide the fillet, but it won't touch a completely separate sketch further down the feature tree), there's this horizontal bar you can drag up and down the feature tree to hide any element after it.

This tutorial might help show what I'm talking about:

[![Youtube: Get the Most Out of SOLIDWORKS FeatureManager Design Tree
](https://img.youtube.com/vi/0iqOAUHxg4E/0.jpg)](https://www.youtube.com/watch?v=0iqOAUHxg4E)

hannobraun, 2023-05-15:
Closing this feature as out of scope. See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

closed 2023-05-15 by hannobraun