hannobraun/fornjot #14: Try to optimize model compile times

hannobraun, 2021-11-23:
I've gone through @mre's excellent [article on faster Rust compile times](https://endler.dev/2020/rust-compile-times/), and I've found a few that might be applicable:

- [Cranelift – The Alternative Rust Compiler](https://endler.dev/2020/rust-compile-times/#cranelift-the-alternative-rust-compiler)
- [Switch To A Faster Linker](https://endler.dev/2020/rust-compile-times/#switch-to-a-faster-linker)
- [Tweak More Codegen Options / Compiler Flags](https://endler.dev/2020/rust-compile-times/#tweak-more-codegen-options-compiler-flags)
- [More Resources](https://endler.dev/2020/rust-compile-times/#more-resources)

There might be more applicable tips in there, but these seems like a good place to start.

freylint, 2022-04-18:
Another possible approach to this could be utilizing a crate like [HVM](https://github.com/Kindelia/HVM)  to create shapes and providing an "code window next to preview window" workflow like openscad.

This would probably also ease development of a traditional CAD workflow as we wouldn't need to embed a full rust toolchain.

hannobraun, 2022-04-19:
Thank you for the suggestion, @freylint. I've seen HVM, but haven't looked into it deeply enough to judge whether it's an option.

Just note that web support is a priority for this project (#73), and for that reason the current plan is to use WebAssembly for the plugin system (#71). In concrete terms, this would probably mean embedding something like [Wasmtime](https://crates.io/crates/wasmtime) on the desktop, and use native browser capabilities to load WASM on the web.

I'd also like to note that Fornjot is modular, so for example the kernel just doesn't care who defined the geometry. It's possible to have alternative frontends that could be based on something else than WASM, although I don't think I would like to maintain those within this repository.

> providing an "code window next to preview window" workflow like openscad.

I'd like to see that in some form. I do think that using an external editor should always have first-class support, as it is likely to be the preferred option for power users. But I do think it would be a huge accessibility advantage if people had everything they need in a single package.

This is one of these areas that are unlikely to see much attention from me for a long while, so any help is appreciated!

> This would probably also ease development of a traditional CAD workflow as we wouldn't need to embed a full rust toolchain.

Yes, absolutely. Rust certainly isn't the ideal option for a CAD modeling language, and the toolchain required is one of the reasons for that. Rust is just the easiest thing to support right now, and I also think there's value in providing an alternative that is so different from the other Code-CAD options.

Switching to WASM will open up the options. I assume that Fornjot will switch to another language as the preferred one at some point; something that is accessible, easy to integrate into a web-based demo, etc. I don't have a clear vision for what that could look like. Previous iterations of Fornjot used [a custom language](https://github.com/hannobraun/kari), but I don't think that's a good option from a cost/benefit perspective.

hannobraun, 2023-03-24:
I just saw this article, which looks light it might be useful: https://rustmagazine.org/issue-2/optimize-rust-comptime/

hannobraun, 2023-05-15:
Closing this issue as out of scope. See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

closed 2023-05-15 by hannobraun