hannobraun/fornjot #17: Show focus point

hannobraun, 2021-12-04:
The focus point is the point on the model (in model coordinates) that the mouse cursor points at. This point is used for input; for example when rotating the model, it is rotated around the focus point.

It would be nice to visualize this point, to show the user that it exists. Otherwise, rotation behavior might be confusing. Making the focus point visible could also help with debugging.

I don't know what approach would work best. Some ideas:

1. Just draw a shape (maybe a circle or triangle) where the focus point is. This would likely not be the best way to visualize it, but it would be easy and help with debugging.
2. Draw a shape, as in 1., but with circle size defined in the model coordinate system. That would help clarify that the shape represents something in the 3D world.
3. Draw a shape, as in 2., but oriented such that its surface normal is parallel to the surface normal of the model at the focus point. This might help clarify that the shape relates to a point on the model.

hannobraun, 2022-06-10:
The input that is affected by the focus point works much better now and has become pretty intuitive (the one caveat being #678, but that should be an easy fix). I don't think showing the focus point would be an advantage, and it might even be more confusing. For this reason, I have decided to close this issue.

Being able to see the focus point would still be an advantage for debugging, and I'd be happy to merge a pull request that adds that (preferably toggleable, like other debug output too).

closed 2022-06-10 by hannobraun

chrisprice, 2022-06-10:
I'm still trying to tidy up my code which I used to debug the other input issues. It's next on my list and I hope to have a PR for this in the next couple of weeks.

hannobraun, 2022-06-10:
That would be awesome. Thank you, @chrisprice!