hannobraun/fornjot #20: When zooming, the camera angle turns towards the focus point. Maybe immediately, maybe there's a limited turning speed, to make it feel less jarring.

hannobraun, 2021-12-07:


hannobraun, 2022-02-21:
I realized that changing the camera direction while zooming is just about, well, *direction*, and has nothing to do with the focus point. Plus, it would be very confusing, if this behavior only worked while the mouse is pointing at the model. I've updated the issue title and description accordingly.

ObiWanRohan, 2022-03-19:
Is someone working on this @hannobraun?

hannobraun, 2022-03-19:
Not to my knowledge, @ObiWanRohan, so this is free to take!

A word of warning though: Something's wrong with input right now. At least rotation sometimes feels very wrong. I think it's just #18, which might not affect this issue. But I also haven't looked into this recently, so yeah, no idea what is actually wrong.

I don't want to discourage you. Any help around input is *very* welcome! I'm just saying, you might be opening a can of worms :grin:

ObiWanRohan, 2022-03-22:
@hannobraun I agree with you that this could just be #18 , which in turn needs #17 to be solved.
I will come back to this one after solving those two issues, since the steps in solving those are going to be needed to fix this issue.

hannobraun, 2022-03-22:
Sounds great :+1:

hannobraun, 2022-04-21:
Hey @ObiWanRohan, I just wanted to note that there's some discussion about larger changes to the graphics/UI aspect going on in #491. This has the potential to disrupt any work that you might be doing on the input side, so if you're still interested in working on that, I suggest you take note.

hannobraun, 2023-05-15:
I've decided to close this issue. `fj-viewer` is only a development tool now, and while improvements to rendering are still in scope and very welcome, I don't think it makes sense to have too many open issues to track potential usability improvements.

See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

closed 2023-05-15 by hannobraun