hannobraun/fornjot #25: Zooming is way too slow

hannobraun, 2021-12-20:
Zooming is way too slow with large models. With the `enclosure` model, zooming basically does nothing.

Right now, there's a given standard zoom speed that is limited by how close the model is, to make viewing details up-close practical. I think this should probably be changed to make zoom speed dependent on distance to the model. Basically, however far away you are, zoom speed should work comfortably, never being too fast or too slow.

If anyone wants to look into this, basically all the relevant code is in [`zoom.rs`](https://github.com/hannobraun/fornjot/blob/cdd7e6e099bce9ff377a28f2ab959bef3c2292e5/src/input/zoom.rs). Specifically the [speed limiting code](https://github.com/hannobraun/fornjot/blob/cdd7e6e099bce9ff377a28f2ab959bef3c2292e5/src/input/zoom.rs#L111-L117) is interesting, and would have to be replaced when fixing this.

hannobraun, 2022-02-14:
As part of addressing this, it would make sense to add a large model the the repository. Doesn't need to be anything special (could be as simple as a cube, or something like that). It would just serve as a test case for input on large models.

eLVas, 2022-05-30:
To be more precise the issue starts to be noticeable with models where at least one side of the bounding box is above 1000.0. And becomes really obvious at around 10000.0

PR #764 (2022-07-06): Simplify zoom and adapt for larger models
This PR removes the interpolated zoom system and replaces it with a basic instantaneous zoom system. The new algorithm moves the camera proportionally based on the current distance from the focus point. This means smaller changes in zoom when close to an object and larger changes in zoom when further away.

Please treat this PR as more of a proposal, as I have I deviated significantly from the initial design.

Close #677.
Close #25.
closed 2022-07-06 by hannobraun