hannobraun/fornjot #31: Begin unit testing

Bandsberg, 2022-01-02:
To be able to systematically narrow down the issue behind #12 I have found that I need start writing unit tests. 
However if it is to early in development to begin doing tests, e.g. most or all will be obsolete in a short period of time, I will focus my energy on other issues for the time being.
@hannobraun are you comfortable with me beginning to include tests?

hannobraun, 2022-01-02:
Hey @Bandsberg, thanks for opening this issue. I'm very comfortable with including unit tests. There used to be tests before (mostly in the core geometry code), but they went away when I replaced said code. I think we've reached the stage, where re-adding unit tests is actually overdue :sweat_smile: 

I also think it makes sense to have integration testing, to make sure I don't accidentally break platform support. I intend to start making plans for this and other things next week. As the code slowly matures, the project should mature too (which includes things like adding a CI build; and me starting to open pull request, like a civilized person, instead of always pushing to `main` directly :grin:).

As far as your specific area of work is concerned, there is a danger that any unit tests might become obsolete as part of implementing #7. Specifically, I currently plan to migrate the plugin system to WebAssembly (to enable web support and for other benefits, as explained in that issue). I expect that this will replace the current approach of dynamic linking completely.

If your concern for wasted effort outweighs your desire for full macOS support, I recommend working on something else for now. I don't know when I'll work on #7 (for now my focus is on core CAD features), but I think it probably won't be more than a few months until I start. (And for the record, I don't insist on implementing #7 myself. Happy to assist any contributor who's interested in that.)

Bandsberg, 2022-01-02:
Great 😃 I will start looking in to this.
Really looking forward to seeing how a plugin system with WebAssembly will look like.

hannobraun, 2022-01-15:
I'm closing this issue. I believe I have answered the question, both in word and in deed (as I've started re-adding unit tests).

Everyone, feel free to still post here, if anything is unclear.

closed 2022-01-15 by hannobraun