hannobraun/fornjot #46: Figure out how to handle experimental features

hannobraun, 2022-01-19:
Fornjot is still very experimental, and it's quite easy for a model to crash the host application. This is not a great user experience, obviously, but trying to be too strict at this early stage (by only accepting really solid code) would stifle progress.

I'd like to carve out a stable subset of features and make any experimental features opt-in somehow, so users know what they're getting into. I'm not yet sure how to do that.

- One option is to add an `experimental` Cargo feature to the `fj` crate. This is not ideal, as models can depend on other models (they are just Cargo packages, after all). One component deep in the dependency tree could enable the `experimental` feature for everyone, making it easy to use experimental features accidentally.
- Maybe it would be better to mark experimental features with an attribute that works somewhat like [`#[deprecated]`](https://doc.rust-lang.org/reference/attributes/diagnostics.html#the-deprecated-attribute). Then users would be warned when using experimental features, or could even add configuration to cause a compiler error in that case. Maybe it's possible to build something like this with a procedural macro. If not, it might be possible to misuse `#[deprecated]` for this purpose.
- Another option is to add some kind of setting to the host application, that will result in models using experimental features to be rejected. I don't think this is viable though. A component might use experimental features in a controlled way that doesn't trigger any misbehavior. A model author might want to use such a component, without enabling use of experimental features in their own code.

If anyone has any ideas, I'd love to hear them! I'm adding the `help wanted` to further encourage feedback.

If this issue doesn't result in a great plan, just adding an `experimental` feature to `fj` should do for a start.

hannobraun, 2022-01-27:
Once this has been implemented, there should be a comment above every `todo!` in the code base, which explains under which experimental feature this code belongs to. Since the presence of a `todo!` would block an experimental feature from being stable, there should be clear documentation which features a `todo!` blocks.

hannobraun, 2022-02-21:
Had the idea to add an `experimental` flag to data structures returned by models. Updated the issue description accordingly.

follower, 2022-03-28:
Random idea that occurs to me about a possible approach:

 * Maybe this could be handled via some (mis?)use of marking experimental code `unsafe` and `deny(unsafe)`/`warn(unsafe)`?

hannobraun, 2022-03-29:
Thank you for the suggestion, @follower. Sounds like a terrible, but very interesting, hack :grin: 

Having thought about this more since my last update here, I've realized that any solutions based on language features won't do, as Fornjot is designed to be language-agnostic, and and some point we will support other modeling languages besides Rust. Given that, I think it's most likely that use of experimental features will be governed in the `fj-operations` crate, and decisions there will be based on flags set in the data structures returned by models.

I'll update this issue when I get a chance.

hannobraun, 2022-04-05:
Since the requirements for this feature are now much clearer than they were initially, I have opened a more specific issue, #431. Closing in favor of that one.

closed 2022-04-05 by hannobraun