hannobraun/fornjot #55: Provide basic user-facing documentation

hannobraun, 2022-01-20:
There should be some basic user-facing documentation. It can be hosted on the website.

Eventually, this should be a comprehensive resource, but for a start, the following information should be enough:
- What is Fornjot?
- How to install it
- How to get started
- Further reading
  - API documentation of `fj` crate
  - Models in repository

Maybe an example of using it together with common tools, like VS Code and Git, would also be good.

ph4ni, 2022-07-26:
I think you should at least have a one-line intro to Fornjot in the about section because that's what people look at as soon as they visit the repo. Example: _The world needs another CAD program. Fornjot is an early-stage project to create a next-generation, code-first Computer-aided design (CAD) application._ (I am using this for the Console interview).

ph4ni, 2022-07-27:
looks much better! Also, can you try adding basic installation instructions? I wanted to try it out but I had no idea where to start.

hannobraun, 2022-07-28:
There's a [*Usage* section in the README](https://github.com/hannobraun/Fornjot#usage), but I see that it's out of date and doesn't cover installation. I'll take a look later!

hannobraun, 2022-07-28:
I've just [submitted a pull request](https://github.com/hannobraun/Fornjot/pull/882) that updates and expands the *Usage* section in the README.

hannobraun, 2023-05-15:
This issue is no longer applicable. See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

closed 2023-05-15 by hannobraun