hannobraun/fornjot #57: Provide Linux/Mac/Windows binaries

hannobraun, 2022-01-20:
To make Fornjot more accessible, and testing easier, we should compile executables for Linux, Mac, and Windows. Since we're already using GitHub Actions for the CI build, it makes sense to use it for this too. This should probably be a separate action that is triggered whenever a normal build on the `main` branch finished. It shouldn't block pull requests.

In the past, I've used the Releases feature to host those binaries. I'd rather reserve Releases for real releases though (as to not confuse users with a million different release, when they click there), so either those binaries should be downloadable through GitHub Actions, or there should be a single nightly/in-progress/WIP release that gets overwritten every time. I don't know if either of those are possible.

Labeling this as `good first issue`, as this only requires knowledge of Cargo and GitHub, not really of anything Fornjot-specific.

hendrikmaus, 2022-02-11:
FYI I am working on this in https://github.com/hannobraun/Fornjot/pull/175

hannobraun, 2022-02-11:
This is mostly done. (Thanks, @hendrikmaus!) There's one remaining issue: https://github.com/hannobraun/Fornjot/pull/175#pullrequestreview-880472207

hendrikmaus, 2022-02-13:
@hannobraun I think we can consider this one done now.

hannobraun, 2022-02-13:
Yes, thank you!

closed 2022-02-13 by hannobraun