hannobraun/fornjot #64: Enforce formatting in CI build

hannobraun, 2022-01-21:
I, like many others, use `cargo fmt` to format my code. To prevent the formatting from drifting, which might cause unrelated formatting changes when someone with formatting enabled edits differently-formatted code again, we should enforce this in the CI build.

Instead of adding a build step for each platform-specific build, I think it would make sense to add a build script and just call that from each build. That reduces duplication and has the advantage of letting developers run the CI build locally, which takes out a lot of guesswork. This build script could then be extended to check formatting.

A shell script won't do, we need something portable. [cargo-make](https://crates.io/crates/cargo-make) looks like it would work.

Labeling as `good first issue`, since no Fornjot knowledge is required. This is just general build system work.

Bandsberg, 2022-01-21:
Any opinion on cargo-make vs nushell?
Nushell goals:

- First and foremost, Nu is cross-platform. Commands and techniques should carry between platforms and offer consistent first-class support for Windows, macOS, and Linux.
- Nu ensures direct compatibility with existing platform-specific executables that make up people's workflows.
- Nu's workflow and tools should have the usability in day-to-day experience of using a shell in 2019 (and beyond).
- Nu views data as both structured and unstructured. It is a structured shell like PowerShell.
- Finally, Nu views data functionally. Rather than using mutation, pipelines act as a means to load, change, and save data without mutable state.

For me the argument of using Nushell would be that it is similar to using most other shells so might easier for more people to understand.
On the other hand (of course) cargo-make is specifically designed as a buildtool for Rust.
Any experiences?

hannobraun, 2022-01-21:
Thanks for the feedback, @Bandsberg!

You make a good case! I didn't really know Nushell (except I remember the name, so I must have seen it at some point).

I haven't used cargo-make before. A friend mentioned it to me recently, and it seemed like a good option. But I'd be happy to try something else, and Nushell does look really good. Definitely more powerful, so it might be more future-proof, if we need to script more stuff in the future (and I'm sure we will).

hannobraun, 2022-02-28:
Addressed in #208.