hannobraun/fornjot #82: Try to speed up CI build

hannobraun, 2022-01-26:
The CI build is slow, mostly because the Windows build takes pretty long. While this is not a critical issue, it would be great if the build was a lot faster.

It should be possible to cache the Cargo output, so all dependencies don't have to be rebuilt all the time. I haven't done that before with GitHub Actions, but there are probably existing solutions, and it should make the build much faster.

mxdamien, 2022-01-31:
Hey !

Caching looks good in my test. Obviously this only starts working with the second run of the CI builds.

[image]

hannobraun, 2022-01-31:
Nice! Love those numbers!

PR #114 (2022-01-31): Use caching to speed up CI build
Closes #82
closed 2022-01-31 by hannobraun