hannobraun/fornjot #122: Investigate using nalgebra in `fj` crate

hannobraun, 2022-02-02:
Right now, the `fj` crate doesn't use a math library, instead just using arrays to represent points and vectors. This works well enough for simple models, but might become inconvenient for more complex ones. I assume that any non-trivial model would benefit from a math library, and having to convert data types into arrays would be an inconvenience.

Using nalgebra in `fj` seems like the obvious choice, but I see two potential problems:

- For technical reasons, all data types in `fj` need to be `#[repr(C)]`. I don't know whether this will cause any problems with nalgabra.
- To keep modeling productive, compile times are critical. I don't know, if having nalgebra as a dependency in `fj` could cause models to compile more slowly.

Especially that second point is hard to judge right now, as all models that are currently available are pretty simple. I do think that using nalgebra types in `fj` makes a lot of sense, but I'd like to see compile times being investigated before adding it.

hannobraun, 2022-02-21:
Due to #193, a lot has changed since I opened this issue. Any investigation should include whether using the `math` module from the host application instead of nalgebra makes sense. This would require the `math` module to be extracted into a separate `fj-math` library (which would be desirable anyway, see #141).

therealprof, 2022-02-23:
Shouldn't the intrinsic types from `math` be exposed for the modeling as well? At the moment a lot of functions are taking raw `f64` or `Vec<[f64; 2]>` which is problematic for a number of reasons including the lack of input validation, pretty printing, handling convenience but also the ability to use make use of other primitive types -- I would certainly enjoy not having to wildly cast around between whatever type I want (or need to) use and `f64` all the time.

Trying to figure out where #246 is actually coming from has not been too much fun, yet. The values are going in as `f64`, then get bounced around and converted a few times internally and come out wrong at the other end. Constructing them early in the model would help quite a lot following along where the incorrect handling is happening.

hannobraun, 2022-02-24:
> Shouldn't the intrinsic types from `math` be exposed for the modeling as well? At the moment a lot of functions are taking raw `f64` or `Vec<[f64; 2]>` which is problematic for a number of reasons [...]

Yes. See my previous comment in this issue.

> Trying to figure out where #246 is actually coming from has not been too much fun, yet. The values are going in as `f64`, then get bounced around and converted a few times internally and come out [...]

Yeah, you're right about that. I'm always careful about including stuff in `fj`, because I'm worried about compile-times. But I agree that the current situation doesn't work too well. We should be using the same math types everywhere.

hannobraun, 2022-02-28:
Updated issue title and description to reflect the current situation.

LoganDark, 2022-05-30:
`nalgebra` is notorious for ridiculous compile times. Have you looked into other crates, such as `ultraviolet`? (I also think the latter would be more suitable to borrow primitives from. See [the documentation](https://docs.rs/ultraviolet) for more details.)

hannobraun, 2022-05-30:
> Is there any particular reason why you couldn't use a sealed trait for this? AFAIK, it's definitely possible to smooth over the different structs using a trait.

That should work. More trait infrastructure could negatively affect compile times though. No idea if it would be an improvement in the end.

Maybe there's a fundamental conflict here: Being generic over dimensions is valuable in the kernel, but probably less so in model code. Maybe it's a mistake to want to use the same library for both use cases, since the constraints are so different. Would be nice to have just one library, of course, but maybe 

> P.S. (slightly offtopic) is there some switch I can flip to make the newsletter send non-tracking links?

No, sorry. I would like to disable link tracking for the whole list, but my mailing list provider (MailerLite) doesn't support that. I contacted support about that a while ago. They acknowledged the feature request, but made no promises on how much of a priority that is going to be.

Didn't seem worth switching providers over this issues, but I'd definitely prefer having no tracking at all in my emails.

LoganDark, 2022-05-30:
> That should work. More trait infrastructure could negatively affect compile times though. No idea if it would be an improvement in the end.

Would certainly be less of an impact than `nalgebra`. The Rust compiler is designed to handle hundreds of different traits per compilation.

> Maybe there's a fundamental conflict here: Being generic over dimensions is valuable in the kernel, but probably less so in model code. Maybe it's a mistake to want to use the same library for both use cases, since the constraints are so different. Would be nice to have just one library, of course, but maybe that's not the best solution.

Perhaps you could, drumroll please, have separate structs and then... use a trait in the kernel to smooth over both of them. :^)

hannobraun, 2022-05-30:
> Would certainly be less of an impact than `nalgebra`. The Rust compiler is designed to handle hundreds of different traits per compilation, but `nalgebra` incurs deep monomorphization costs (tons of codegen).

Good to know! I have to admit, I don't have great intuition about what kind of feature is going to incur how much of a compile-time cost.

> Perhaps you could, drumroll please, have separate structs and then... use a trait in the kernel to smooth over both of them. :^)

I think that would be roughly the same as using two different libraries. And if we decided that having those smoothing-over-traits *only* available in the kernel would not be enough, it would turn into *exactly* the same as using two libraries :smile: 

---

Your suggestion about using `ultraviolet` gave me a thought: Why not use [`mint`](https://crates.io/crates/mint) in the `fj` crate? That could be much more lightweight, and would allow users to choose whatever math library they want to, as long as that library supports `mint`, which most (all?) of them seem to do.

LoganDark, 2022-05-30:
> I think that would be roughly the same as using two different libraries. And if we decided that having those smoothing-over-traits _only_ available in the kernel would not be enough, it would turn into _exactly_ the same as using two libraries 😄

Well, if you put the traits in the main math library with the rest of the stuff, it would turn into having *one* library.

> Your suggestion about using `ultraviolet` gave me a thought: Why not use [`mint`](https://crates.io/crates/mint) in the `fj` crate? That could be much more lightweight, and would allow users to [...]

I don't have any experience with `mint`, but that sounds like it could work.

LoganDark, 2022-06-14:
P.S. did you actually notice `cint` is made by the same dev who made `ultraviolet`? I didn't actually realize until now but it's actually kind of ironic...

LoganDark, 2022-06-14:
OOPS I meant `mint`. It's `mint` that's made by the same author. I keep getting them confused because they're only one letter off...

hannobraun, 2022-06-14:
I don't think that's right? According to crates.io, `ultraviolet` and `cint` are by Gray Olson, while `mint` is by Dzmitry Malyshau.

hannobraun, 2023-03-02:
I've had an idea: We could move the math types into a new crate, say `fj-core`, while leaving the math *operations* where they are in `fj-math`. So just the type definitions (and maybe some simple methods like basic constructors and accessors) would move to the new library. Anything non-trivial would stay in `fj-math`, which would implement the operations via extension traits.

As a result, the kernel could keep using the math types with minimal change, but the `fj` crate could now depend on the new `fj-core` (or whatever it ends up being called) and get the math types, without incurring too much of a compile-time cost.

By itself, this would be an incremental improvement. The math types are not any worse than using arrays for vectors and such, while reducing friction between models and kernel. Even increasing reliability, by using [`Scalar`](https://docs.rs/fj-math/0.38.0/fj_math/struct.Scalar.html). It wouldn't be a long-term solution though, as having vector operations and other math stuff in models is still desirable.

However, it may be possible to come up with some trait-based proxy approach, so models could call the math operations through the FFI boundary, without having to compile them. I'm not sure about that, as it's just a vague idea at this point. But if that's possible to do in a relatively low-friction way, there's the potential to use this technique for many other APIs beyond just math.

hannobraun, 2023-05-15:
This issue is no longer applicable. The affected code [has been removed](https://github.com/hannobraun/fornjot/pull/1822). See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.