hannobraun/fornjot #143: Star model displays triangulation issues

hannobraun, 2022-02-07:
When changing the number of pointy ends of the star model, some values display triangulation issues.

Here's the 12-pointed star:
[image]

And here's the 15-pointed star:
[image]

These might or might not be the result of issues that are already documented in #105 or #138, but in any case, I think it would be a mistake to fix more triangulation bugs without adding test cases to guard against regressions. Labeling this issue as blocked by #105 and #138.

hannobraun, 2022-02-14:
#138 has been addressed, but the problems described here are still present. This means this only blocked on #105 now. Updated the issue description accordingly.

hannobraun, 2022-02-24:
@therealprof has reported some weird data in #246. See also my comment there.

I'm going to repeat some of what I said there here, as it directly relates to this issue.

---

The screenshots here show whole triangles that shouldn't be there. This is most likely a problem in the [triangulation code](https://github.com/hannobraun/Fornjot/blob/a774cc94fb063d062074fee68a8e3a1f4c1c078e/src/kernel/topology/faces.rs#L98-L234) or the [approximation code](https://github.com/hannobraun/Fornjot/blob/a774cc94fb063d062074fee68a8e3a1f4c1c078e/src/kernel/approximation.rs) that feeds it data. More specifically, I suspect that it might be a problem with the point-in-polygon testing that's necessary as part of the triangulation ([[1]](https://github.com/hannobraun/Fornjot/blob/a774cc94fb063d062074fee68a8e3a1f4c1c078e/src/kernel/topology/faces.rs#L151-L158) and [[2]](https://github.com/hannobraun/Fornjot/blob/a774cc94fb063d062074fee68a8e3a1f4c1c078e/src/kernel/topology/faces.rs#L185-L212)), which has known issues (#105).

A word of warning: That triangulation code is pretty gnarly, and overdue for a cleanup. See #105.

therealprof, 2022-02-24:
I've trimmed down the example a bit to only render the "outer" sketch so there're less datapoints (and involved code) to look at...

Problems start at 12 "points", here's what that looks like:

[image]

hannobraun, 2022-02-24:
Looks like triangles that should be kept are being removed instead, which again points to issues with the point-in-polygon test.

hannobraun, 2022-04-09:
Even the default version of the star model is broken now:
[image]

Most likely, this happened yesterday in #439.

I'm looking into it. #439 was intended to be a pure refactoring without any functional changes.

hannobraun, 2022-04-09:
I think I get it. The problem was introduced in this commit: https://github.com/hannobraun/Fornjot/pull/439/commits/eb187e033ea696b9b4d08060ff800f92748693d9

This changes the calculation of the point the ray shoots to. The algorithm itself stays the same, just the ray that is used for determining if a triangle is part of the shape is changed, meaning it could hit different edges in different places, at different angles.

There are known problem with that code. I'm very sure that this commit didn't introduce any new issues, it just shuffled things around, so that the default 5-pointed star model now triggers those bugs. This is unfortunate, but some fallout is to be expected, as I work on the triangulation code.

I don't intend to do anything about this specific instance of the bug, but I'm actively working on fixing this and other triangulation bugs, so I hope this will improve soon.

PR #448 (2022-04-09): Fix triangulation issues regarding point-in-polygon test
This pull request mostly rewrites the point-in-polygon test, adding a test suite to provide some assurance. The point-in-polygon test was responsible for one class of triangulation bugs that seems to have cause pretty wide-spread issues.

Close #105 
Close #143
closed 2022-04-09 by hannobraun