hannobraun/fornjot #154: Export to STL

hannobraun, 2022-02-09:
Fornjot already supports exporting to 3MF. Since STL is also based on triangle meshes, adding support should be fairly straight-forward and not require many changes to existing Fornjot code.

The [export code currently lives in `main.rs`](https://github.com/hannobraun/Fornjot/blob/bf8fe8d05c9b5910366771d76d6890c53d903bad/src/main.rs#L94-L126). It can be extended to check the file ending of the export path, and choose the file format accordingly. I think this would also be a good opportunity to clean up `main.rs` a bit, and move that export code to a separate module.

There might already be libraries for writing STL files that we can use. I haven't looked into that.

Labeling as https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this doesn't require much knowledge of Fornjot, and might be a good introduction to it.

chrisprice, 2022-05-12:
From a quick hunt around crates.io, there seem to be two popular crates for working with STL files -

* `stl_io` - [GitHub](https://github.com/hmeyer/stl_io) [Crates.io](https://crates.io/crates/stl_io)
* `nom_stl` - [GitHub](https://github.com/fast-radius/nom_stl) [Crates.io](https://crates.io/crates/nom_stl)

I'm a little confused/concerned by the license of `stl_io`. According to the `Cargo.toml` it is GPL-3.0, according to LICENSE it is GPL-3.0 and according to the README.md it is dual licensed Apache 2.0 and MIT. A [year old PR](https://github.com/hmeyer/stl_io/pull/14) which attempts to address this has yet to be accepted. Conversely `nom_stl` is consistently MIT licensed.

However, `nom_stl` is currently only able to read/parse STL files and not write/serialize them so I went with `stl_io`. Is the license an issue for use in this project?

I've knocked together a [branch](https://github.com/hannobraun/Fornjot/compare/main...chrisprice:support-emf) with a first pass at an implementation and I was pleasantly surprised to find that macOS natively supports STL files in Preview -

[image]

I'd still like to investigate exporting to an [IndexedMesh](https://docs.rs/stl_io/0.6.0/stl_io/struct.IndexedMesh.html) as I think it might be a simpler approach (I've done no research to back this up).

The files in the preview above were generated by running the below test script -
```sh
#!/bin/sh

cargo run -- --model cuboid --export cuboid.stl
cargo run -- --model cuboid --export cuboid.3mf
cargo run -- --model cuboid --export cuboid.unknown
cargo run -- --model cuboid --export cuboid

cargo run -- --model group --export group.stl
cargo run -- --model group --export group.3mf
cargo run -- --model group --export group.unknown
cargo run -- --model group --export group

cargo run -- --model spacer --export spacer.stl
cargo run -- --model spacer --export spacer.3mf
cargo run -- --model spacer --export spacer.unknown
cargo run -- --model spacer --export spacer

cargo run -- --model star --export star.stl
cargo run -- --model star --export star.3mf
cargo run -- --model star --export star.unknown
cargo run -- --model star --export star

```

Which produces the following output -

```
    Finished dev [unoptimized + debuginfo] target(s) in 6.54s
     Running `target/debug/fj-app --model cuboid --export cuboid.stl`
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
    Finished dev [unoptimized + debuginfo] target(s) in 0.33s
     Running `target/debug/fj-app --model cuboid --export cuboid.3mf`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
    Finished dev [unoptimized + debuginfo] target(s) in 0.23s
     Running `target/debug/fj-app --model cuboid --export cuboid.unknown`
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
Error: Extension not recognised, got "unknown"
    Finished dev [unoptimized + debuginfo] target(s) in 0.23s
     Running `target/debug/fj-app --model cuboid --export cuboid`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
Error: No extension specified
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target/debug/fj-app --model group --export group.stl`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
    Finished dev [unoptimized + debuginfo] target(s) in 0.22s
     Running `target/debug/fj-app --model group --export group.3mf`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target/debug/fj-app --model group --export group.unknown`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
Error: Extension not recognised, got "unknown"
    Finished dev [unoptimized + debuginfo] target(s) in 0.25s
     Running `target/debug/fj-app --model group --export group`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
Error: No extension specified
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target/debug/fj-app --model spacer --export spacer.stl`
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target/debug/fj-app --model spacer --export spacer.3mf`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
    Finished dev [unoptimized + debuginfo] target(s) in 0.26s
     Running `target/debug/fj-app --model spacer --export spacer.unknown`
    Finished dev [unoptimized + debuginfo] target(s) in 0.06s
Error: Extension not recognised, got "unknown"
    Finished dev [unoptimized + debuginfo] target(s) in 0.25s
     Running `target/debug/fj-app --model spacer --export spacer`
    Finished dev [unoptimized + debuginfo] target(s) in 0.06s
Error: No extension specified
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target/debug/fj-app --model star --export star.stl`
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
    Finished dev [unoptimized + debuginfo] target(s) in 0.22s
     Running `target/debug/fj-app --model star --export star.3mf`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
    Finished dev [unoptimized + debuginfo] target(s) in 0.23s
     Running `target/debug/fj-app --model star --export star.unknown`
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
Error: Extension not recognised, got "unknown"
    Finished dev [unoptimized + debuginfo] 
target(s) in 0.23s
     Running `target/debug/fj-app --model star --export star`
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
Error: No extension specified
```

[Attached](https://github.com/hannobraun/Fornjot/files/8682321/Archive.zip) are the STL files.

hannobraun, 2022-05-13:
Thanks for looking into this, @chrisprice!

> I'm a little confused/concerned by the license of `stl_io`. According to the `Cargo.toml` it is GPL-3.0, according to LICENSE it is GPL-3.0 and according to the README.md it is dual licensed Apache [...]

That's an unfortunate situation. The license would indeed be an issue. Fornjot is licensed under 0BSD, basically the most permissive license there is, and that's deliberate. Using a GPL-licensed library would completely change the requirements we put on people that want to use Fornjot libraries, or distribute anything from Fornjot.

---

I just did a quick search and fount `stl` ([crates.io](https://crates.io/crates/stl), [GitHub](https://github.com/eholk/rust-stl)). Seems to support reading and writing, and seems to be better maintained (more recent activity, no outstanding pull requests). Maybe that will work for us.

If, for some reasons, it doesn't, I suspect that rolling our own STL export would be practical as well. I haven't looked into STL in detail, but from what I know, it's a simple format. And for writing, we wouldn't even need to support all of it, just the subset we need. 3MF [wasn't much of a problem](https://github.com/hannobraun/3mf-rs), after all.

chrisprice, 2022-05-16:
I skipped over `stl` originally because of the low download count. However, I really hadn't appreciated just how simple the STL format was. The implementation of `stl` is very straightforward and as you say if it ever needs replacing then rolling our own would be fine.

I've created a PR covering the `stl` implementation.

PR #594 (2022-05-17): Support emf
Fixes #154
closed 2022-05-17 by hannobraun