hannobraun/fornjot #157: Fix wgpu warnings

hannobraun, 2022-02-09:
Since I re-enabled logging recently, I've been noticing warnings being logged by wgpu. Here's the output from a typical run:

```
2022-02-09T13:27:48.037289Z  WARN wgpu_hal::vulkan::instance: Unable to find layer: VK_LAYER_KHRONOS_validation    
2022-02-09T13:27:48.039542Z  INFO wgpu_hal::vulkan::instance: Instance version: 0x4020cb    
2022-02-09T13:27:48.039672Z  INFO wgpu_hal::vulkan::instance: Enabling debug utils    
2022-02-09T13:27:48.039793Z  INFO wgpu_hal::vulkan::instance: Enabling device properties2    
MESA-INTEL: warning: Haswell Vulkan support is incomplete
2022-02-09T13:27:48.045609Z  INFO wgpu_core::instance: Adapter Vulkan AdapterInfo { name: "Intel(R) HD Graphics 4600 (HSW GT2)", vendor: 32902, device: 1042, device_type: IntegratedGpu, backend: Vulkan }    
2022-02-09T13:27:48.076110Z  INFO wgpu_hal::vulkan::adapter: Private capabilities: PrivateCapabilities { flip_y_requires_shift: true, imageless_framebuffers: true, image_view_usage: true, timeline_semaphores: true, texture_d24: true, texture_d24_s8: true, can_present: true, non_coherent_map_mask: 63, robust_buffer_access: true, robust_image_access: true }    
2022-02-09T13:27:48.079807Z  INFO wgpu_core::device: configuring surface with SurfaceConfiguration { usage: RENDER_ATTACHMENT, format: Bgra8UnormSrgb, width: 800, height: 600, present_mode: Fifo }    
2022-02-09T13:27:48.082956Z  INFO wgpu_core::device: Created texture Valid((0, 1, Vulkan)) with TextureDescriptor { label: None, size: Extent3d { width: 800, height: 600, depth_or_array_layers: 1 }, mip_level_count: 1, sample_count: 1, dimension: D2, format: Depth32Float, usage: RENDER_ATTACHMENT }    
2022-02-09T13:27:48.090101Z  INFO wgpu_core::device: Created buffer Valid((0, 1, Vulkan)) with BufferDescriptor { label: None, size: 128, usage: COPY_DST | UNIFORM, mapped_at_creation: true }    
2022-02-09T13:27:48.094594Z  INFO wgpu_core::device: Created buffer Valid((1, 1, Vulkan)) with BufferDescriptor { label: None, size: 0, usage: VERTEX, mapped_at_creation: false }    
2022-02-09T13:27:48.094776Z  INFO wgpu_core::device: Created buffer Valid((2, 1, Vulkan)) with BufferDescriptor { label: None, size: 0, usage: INDEX, mapped_at_creation: false }    
2022-02-09T13:27:48.094888Z  INFO wgpu_core::device: Created buffer Valid((3, 1, Vulkan)) with BufferDescriptor { label: None, size: 0, usage: VERTEX, mapped_at_creation: false }    
2022-02-09T13:27:48.095030Z  INFO wgpu_core::device: Created buffer Valid((4, 1, Vulkan)) with BufferDescriptor { label: None, size: 0, usage: INDEX, mapped_at_creation: false }    
2022-02-09T13:27:48.107642Z  INFO wgpu_core::device: Created buffer Valid((5, 1, Vulkan)) with BufferDescriptor { label: None, size: 64, usage: COPY_DST | UNIFORM, mapped_at_creation: true }    
2022-02-09T13:27:48.108572Z  INFO wgpu_core::device: Created texture Valid((1, 1, Vulkan)) with TextureDescriptor { label: Some("wgpu_glyph::Cache"), size: Extent3d { width: 256, height: 256, depth_or_array_layers: 1 }, mip_level_count: 1, sample_count: 1, dimension: D2, format: R8Unorm, usage: COPY_DST | TEXTURE_BINDING }    
2022-02-09T13:27:48.108923Z  INFO wgpu_core::device: Created buffer Valid((6, 1, Vulkan)) with BufferDescriptor { label: Some("wgpu_glyph::Cache upload buffer"), size: 25600, usage: COPY_SRC | COPY_DST, mapped_at_creation: false }    
2022-02-09T13:27:48.110579Z  INFO wgpu_core::device: Created buffer Valid((7, 1, Vulkan)) with BufferDescriptor { label: Some("wgpu_glyph::Pipeline instances"), size: 2600000, usage: COPY_DST | VERTEX, mapped_at_creation: false }    
2022-02-09T13:27:48.144042Z  INFO wgpu_core::device: Created buffer Valid((8, 1, Vulkan)) with BufferDescriptor { label: None, size: 29040, usage: VERTEX, mapped_at_creation: true }    
2022-02-09T13:27:48.144858Z  INFO wgpu_core::device: Created buffer Valid((9, 1, Vulkan)) with BufferDescriptor { label: None, size: 5808, usage: INDEX, mapped_at_creation: true }    
2022-02-09T13:27:48.145553Z  INFO wgpu_core::device: Created buffer Valid((10, 1, Vulkan)) with BufferDescriptor { label: None, size: 198080, usage: VERTEX, mapped_at_creation: true }    
2022-02-09T13:27:48.146281Z  INFO wgpu_core::device: Created buffer Valid((11, 1, Vulkan)) with BufferDescriptor { label: None, size: 19808, usage: INDEX, mapped_at_creation: true }    
2022-02-09T13:27:48.146662Z  INFO wgpu_core::device: Buffer (1, 1, Vulkan) is dropped    
2022-02-09T13:27:48.146819Z  INFO wgpu_core::device: Buffer (2, 1, Vulkan) is dropped    
2022-02-09T13:27:48.146936Z  INFO wgpu_core::device: Buffer (3, 1, Vulkan) is dropped    
2022-02-09T13:27:48.147039Z  INFO wgpu_core::device: Buffer (4, 1, Vulkan) is dropped    
2022-02-09T13:27:48.154246Z  INFO wgpu_core::device: configuring surface with SurfaceConfiguration { usage: RENDER_ATTACHMENT, format: Bgra8UnormSrgb, width: 1912, height: 1016, present_mode: Fifo }    
2022-02-09T13:27:48.169484Z  INFO wgpu_core::device: Created texture Valid((2, 1, Vulkan)) with TextureDescriptor { label: None, size: Extent3d { width: 1912, height: 1016, depth_or_array_layers: 1 }, mip_level_count: 1, sample_count: 1, dimension: D2, format: Depth32Float, usage: RENDER_ATTACHMENT }    
2022-02-09T13:27:48.200567Z  INFO wgpu_core::device: Created buffer Valid((12, 1, Vulkan)) with BufferDescriptor { label: Some("wgpu_glyph::Cache upload buffer"), size: 59136, usage: COPY_SRC | COPY_DST, mapped_at_creation: false }    
2022-02-09T13:27:48.200733Z  INFO wgpu_core::device: Buffer (6, 1, Vulkan) is dropped    
2022-02-09T13:27:48.204066Z  INFO wgpu_core::device: Created buffer Valid((13, 1, Vulkan)) with BufferDescriptor { label: Some("staging"), size: 59136, usage: MAP_WRITE | COPY_SRC, mapped_at_creation: true }    
2022-02-09T13:27:48.205754Z  INFO wgpu_core::device: Created buffer Valid((14, 1, Vulkan)) with BufferDescriptor { label: Some("staging"), size: 5876, usage: MAP_WRITE | COPY_SRC, mapped_at_creation: true }    
2022-02-09T13:27:48.208605Z  INFO wgpu_core::device: Created buffer Valid((15, 1, Vulkan)) with BufferDescriptor { label: Some("staging"), size: 1024, usage: MAP_WRITE | COPY_SRC, mapped_at_creation: true }    
2022-02-09T13:27:48.210105Z  INFO wgpu_core::device: Buffer (13, 1, Vulkan) is dropped    
2022-02-09T13:27:48.210209Z  INFO wgpu_core::device: Buffer (14, 1, Vulkan) is dropped    
2022-02-09T13:27:48.210280Z  INFO wgpu_core::device: Buffer (15, 1, Vulkan) is dropped    
2022-02-09T13:27:48.210514Z  WARN wgpu_core::device::queue: Dropped buffer has a pending mapping.    
2022-02-09T13:27:48.210572Z  WARN wgpu_core::device::queue: Dropped buffer has a pending mapping.    
2022-02-09T13:27:48.210619Z  WARN wgpu_core::device::queue: Dropped buffer has a pending mapping.    
2022-02-09T13:27:48.307232Z  INFO wgpu_core::device: configuring surface with SurfaceConfiguration { usage: RENDER_ATTACHMENT, format: Bgra8UnormSrgb, width: 3824, height: 2032, present_mode: Fifo }    
2022-02-09T13:27:48.339327Z  INFO wgpu_core::device: Created texture Valid((3, 3, Vulkan)) with TextureDescriptor { label: None, size: Extent3d { width: 3824, height: 2032, depth_or_array_layers: 1 }, mip_level_count: 1, sample_count: 1, dimension: D2, format: Depth32Float, usage: RENDER_ATTACHMENT }    
2022-02-09T13:27:48.345226Z  INFO wgpu_core::device: Created buffer Valid((15, 2, Vulkan)) with BufferDescriptor { label: Some("staging"), size: 1024, usage: MAP_WRITE | COPY_SRC, mapped_at_creation: true }    
2022-02-09T13:27:48.348008Z  INFO wgpu_core::device: Buffer (15, 2, Vulkan) is dropped    
2022-02-09T13:27:48.348669Z  WARN wgpu_core::device::queue: Dropped buffer has a pending mapping.    
2022-02-09T13:27:51.493964Z  INFO wgpu_core::device: Buffer (0, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494043Z  INFO wgpu_core::device: Buffer (8, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494093Z  INFO wgpu_core::device: Buffer (9, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494138Z  INFO wgpu_core::device: Buffer (10, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494177Z  INFO wgpu_core::device: Buffer (11, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494256Z  INFO wgpu_core::device: Buffer (5, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494331Z  INFO wgpu_core::device: Buffer (12, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494401Z  INFO wgpu_core::device: Buffer (7, 1, Vulkan) is dropped    
2022-02-09T13:27:51.494443Z  INFO wgpu_core::hub: Dropping Global    
2022-02-09T13:27:51.524497Z  INFO wgpu_core::device: Destroying 3 command encoders    
```

I don't know if this is causing any problems, but it's certainly worth looking into, to make sure.

liubog2008, 2022-03-24:
The first WARN is missing validation layer of vulkan. Just install the SDK of vulkan.

In archlinux, `vulkan-devel` is what you need.

hannobraun, 2022-03-24:
That indeed worked, thank you @liubog2008!

Together with the changes in #397, that fixes all warnings on my machine. Closing.

closed 2022-03-24 by hannobraun