hannobraun/fornjot #223: Useful resource

giannissc, 2022-02-21:
This is an exciting project and I hope it succeds! Here are some resource you might find useful:
(I am not an expert in any of these. I was just curious at some point and I found this to explain somee of the stuff I was curious about but it might not be that useuful in the end.)

Boundary representation:
Solid Modelling and CAD Systems by Ian Stroud
Applied Geometry for Computer Graphics and CAD by Duncan Marsh
Boundary Representation Modelling Techniques by Ian Stroud

Contraint modelling:
[1]: [Interactive 2D Constraint-Based Geometric Construction System](https://github.com/GraphiteEditor/Graphite/files/6676933/41d4.content.pdf)
[2]: [Geometric Constraint Solving in Parametric Computer-Aided Design](https://github.com/GraphiteEditor/Graphite/files/6676971/021001_1.pdf)
[3]: [Constraint-based solid modeling with geometric features Zhang, J.,](https://github.com/GraphiteEditor/Graphite/files/6677016/134.pdf)
[4]: [A geometric constraint engine](https://github.com/GraphiteEditor/Graphite/files/6676979/1-s2.0-000437029290012M-main.pdf)
[5]: [Using Degrees of Freedom Analysis To Solve Geometric Constraint Systems](https://github.com/GraphiteEditor/Graphite/files/6676981/112515.112566.pdf)
[6]: [Solving Geometric Constraint Systems](https://github.com/GraphiteEditor/Graphite/files/6676982/AAAI90-106.pdf)
[7]: [A Graph-constructive Approach to Solving Systems of Geometric Constraints](https://github.com/GraphiteEditor/Graphite/files/6677010/TOG97.pdf)

hannobraun, 2022-02-21:
Thank you, @giannissc, this is very helpful!

> Boundary Representation Modelling Techniques by Ian Stroud

I'm currently reading through this one. It's already been really helpful, so I'm looking forward to checking out the other ones.

---

Now I need to figure out where to put these links, so I don't forget them by the time I need them. I guess just keeping this issue open will do for now, while I figure something better.

giannissc, 2022-02-21:
You might also find the work of matt keeter interesting as well!
https://www.mattkeeter.com/projects/mrep/
https://www.mattkeeter.com/projects/foxtrot/

hannobraun, 2022-02-21:
> Maybe you can keep the issue here to put all resources relevant to cad modelling or create a new repo just for resources

I like to use issues to track actionable work items. Using them to track too much other stuff makes it harder to keep an overview, which is why I'd like to find a more appropriate place.

But while I do that, one open issue to collect resources doesn't hurt, so keep them coming :smile:

follower, 2022-03-28:
Another endorsement here for the suggestion of looking at Matt Keeter's work. :)

While my ability to understand the underlying mathematics/algorithm work is limited, :D I do see a _lot_ of potential in the UX of "(node and/or code) + direct manipulation + visual feedback" as a paradigm.

This paradigm is something Matt has explored via a project progression that includes:

 1. [Antimony](https://www.mattkeeter.com/projects/antimony/3/)
 2. [libfive Studio](https://libfive.com/studio/)

So might be interesting work to look at  when Fornjot is at a stage to be focused more on the model creation UX aspect.

hannobraun, 2022-03-29:
Thank you for the feedback, @follower!

I've used libfive Studio, but I have yet to take a look at Antimony. I'll have to check it out!