hannobraun/fornjot #269: Add local build script

hannobraun, 2022-03-01:
We have a comprehensive CI build now that checks quite a few things. This is good and necessary, but I know from experience, that it can also be a pain for new contributors. You don't know what the CI build checks for yet, so you don't know what to pay attention to. Since CI builds are not instant, it can be a pain to fix your broken build.

Experienced contributors also benefit from an easy way to check the most relevant things locally, before pushing a pull request.

Ideally, this local build would not duplicate what the CI build already does, but due to the way the CI build is structured, this is not practical. And it doesn't matter much, as there are only a few things that are checked. Duplicating those in a local build script is totally fine.

Here's what this build script should do:
- `cargo clippy`
- `cargo test`
- `cargo fmt --check`

This is rather straight-forward. The question is, what should this build script be, as it needs to work on many platforms. @Bandsberg suggested [Nushell](https://www.nushell.sh/) in https://github.com/hannobraun/Fornjot/issues/64#issuecomment-1018751658, but there are many possibilities.

Labeling https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this is rather self-contained and doesn't require knowledge of Fornjot.

chrisprice, 2022-05-09:
Would an implementation using [`just`](https://just.systems/) be appropriate?

I quickly threw [this branch](https://github.com/hannobraun/Fornjot/compare/main...chrisprice:add-local-build) together as a proof of concept but I'm more than happy to use something else if there's a preference.

hannobraun, 2022-05-10:
> Would an implementation using [`just`](https://just.systems/) be appropriate?

Yes, I think it would. I've used in on another project in the past and never had complaints. Plus, it's still actively maintained.

> I quickly threw [this branch](https://github.com/hannobraun/Fornjot/compare/main...chrisprice:add-local-build) together as a proof of concept

Looks great! The only thing I could think to add is a note, that the `justfile` is maintained in parallel to the CI configuration, that it's possible they can deviate, and that most CI failures that aren't caught by `just build` should be considered bugs in the `justfile`.

> but I'm more than happy to use something else if there's a preference.

Personally, I was excited to try out Nushell, but I don't think it's a better solution to this problem.

PR #569 (2022-05-11): Add local build
Fixes #269.
closed 2022-05-11 by hannobraun