hannobraun/fornjot #271: Check documentation in CI build

hannobraun, 2022-03-01:
The rustdoc documentation had accumulated a few warnings that I fixed (https://github.com/hannobraun/Fornjot/pull/270). This should be checked in the CI build. A check can be added to [`workflows/test.yml`](https://github.com/hannobraun/Fornjot/blob/main/.github/workflows/test.yml).

Labeling https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this is a self-contained change that doesn't require knowledge of Fornjot.

RishiKumarRay, 2022-03-03:
can you tell me any references used for checking documentation of rust i can integrating that

hannobraun, 2022-03-03:
Hey @RishiKumarRay, thank you for your interest!

I'm not 100% sure myself how to do this, but here's what I know:
1. You can build the API documentation by running `cargo doc`. This will emit warnings, if anything is wrong in the documentation.
2. You can see in [`workflows/test.yml`](https://github.com/hannobraun/Fornjot/blob/main/.github/workflows/test.yml) how to run Cargo commands in our CI build. We don't need to run this on multiple platforms, so copying the "Format" or "Clippy Check"  jobs should work.
3. To make that job fail when `cargo doc` emits a warning, I think we can set the `RUSTDOCFLAGS="-D warnings"` environment variable. I don't know how to set environment variables in GitHub Actions, but I'm sure it's possible somehow.

Does that help?

RishiKumarRay, 2022-03-06:
@hannobraun  Do you think this is right?
 Doc
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@ec3a7ce113134d7a93b817d10a8272cb61118579
      - uses: actions-rs/toolchain@568dc894a7f9e32ffd9bb7d7a6cebb784cdaa2b0
        with:
          toolchain: stable
          components: rustDoc
          profile: minimal
      - uses: actions-rs/cargo@4ff6ec2846f6e7217c1a9b0b503506665f134c4b
        with:
          command: cargo doc
          args: RUSTDOCFLAGS="-D warnings"

hannobraun, 2022-04-19:
Looks like more warnings have crept in again: #490 (thanks for letting us know, @freylint!)

PR #766 (2022-07-04): Check for docs warnings in CI
Close #271
closed 2022-07-04 by hannobraun