hannobraun/fornjot #306: CI configuration doesn't work well with non-code changes

hannobraun, 2022-03-08:
In [`workflows/test.yml`], there's this piece of configuration:

``` yaml
paths-ignore:
  - ".vscode/**"
  - "**.md"
```

This makes sense, but it does have an unintended side effect: When I submit a pull request to make changes to the README, for example, the workflow doesn't run at all. But some of the jobs are marked as required for merging in the repository configuration, meaning my [not so hypothetical](https://github.com/hannobraun/Fornjot/pull/305) pull request can't be merged without overriding the configuration.

Overriding the configuration every time I make such a change is inconvenient. Disabling this configuration permanently is undesirable, as it allows me to use the auto-merge feature, which is super-convenient.

It would be ideal, if those jobs could report success in such a case, without having run at all. I don't know, if that's possible. @hendrikmaus, @ObiWanRohan, any thoughts?

Labeling as https://github.com/hannobraun/Fornjot/labels/help%20wanted, as this build stuff is a bit out of my wheelhouse.

hannobraun, 2022-03-08:
Update: Turns out I can't quickly disable and re-enable the required configuration. I have to re-select every single check that's required, which is extra inconvenient and error-prone.

I think if we can't come up with a good solution, I'll remove that `paths-ignore` bit before my next non-code pull request.

hendrikmaus, 2022-03-08:
Now that you mention it, I recently ran into the same and removed the `paths-ignore` portion as well.

hannobraun, 2022-03-09:
I'm about to merge a Markdown-only change, and since I don't want to go through the configuration change again, I've just removed `paths-ignore` in #314. If anyone has a better idea, I'm happy to hear it, but otherwise I consider this problem solved.

closed 2022-03-09 by hannobraun

hendrikmaus, 2022-03-10:
GitHub has an "official" solution which I find ridiculous: https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/defining-the-mergeability-of-pull-requests/troubleshooting-required-status-checks#handling-skipped-but-required-checks

tl;dr create a workflow that covers only the files the actual workflow ignores and have it always pass.

hannobraun, 2022-03-10:
Thanks for looking into it, @hendrikmaus.

Seems like more trouble than it's worth. It's probably not a disaster, to just keep running all CI builds on every pull request. The vast majority are code changes anyway.

commit 9f25f779 (2022-03-09): Trigger CI build for all paths

Otherwise it becomes impossible to merge a non-code change without
changing the branch protection configuration of the repository.

Close #306
PR #316 (2022-03-09): Really remove `paths-ignore` config in CI
Turns out I missed a spot previously. This should really fix #306 now.
commit 70b96868 (2022-03-09): Really remove `paths-ignore` config in CI

Turns out I missed a spot previously. This should really fix #306 now.