hannobraun/fornjot #324: Issues when firing up project on a Nvidia GPU

ramboldio, 2022-03-09:
I started the project up on a Dell XPS 15 (nvidia gpu and a inactive intel uhd) on ubuntu with wayland and ran into this issue:

 
```
(base) ➜  Fornjot git:(main) RUST_BACKTRACE=1 cargo run -- -m spacer
Finished dev [unoptimized + debuginfo] target(s) in 1.24s
     Running `target/debug/fj-host -m spacer`
    Finished dev [unoptimized + debuginfo] target(s) in 0.00s
  2022-03-09T21:43:14.867820Z  WARN wgpu_hal::vulkan::instance: Unable to find layer: VK_LAYER_KHRONOS_validation
    at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/wgpu-hal-0.12.4/src/vulkan/[instance.rs:512](http://instance.rs:512/)

MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

  2022-03-09T21:43:14.884583Z  WARN wgpu_hal::vulkan::instance: Disabling presentation on 'Intel(R) UHD Graphics (CML GT2)' (id 0x56532c9cdaa0) because of NV Optimus (on Linux)
    at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/wgpu-hal-0.12.4/src/vulkan/[instance.rs:629](http://instance.rs:629/)

  2022-03-09T21:43:14.925799Z  WARN wgpu_hal::vulkan: Unrecognized device error ERROR_INITIALIZATION_FAILED
    at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/wgpu-hal-0.12.4/src/vulkan/[mod.rs:675](http://mod.rs:675/)

thread 'main' panicked at 'Error in Surface::configure: parent device is lost', /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/wgpu-0.12.0/src/backend/[direct.rs:214](http://direct.rs:214/):9
stack backtrace:
   0: rust_begin_unwind
             at /rustc/db9d1b20bba1968c1ec1fc49616d4742c1725b4b/library/std/src/[panicking.rs:498](http://panicking.rs:498/):5
   1: core::panicking::panic_fmt
             at /rustc/db9d1b20bba1968c1ec1fc49616d4742c1725b4b/library/core/src/[panicking.rs:107](http://panicking.rs:107/):14
   2: wgpu::backend::direct::Context::handle_error_fatal
             at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/wgpu-0.12.0/src/backend/[direct.rs:214](http://direct.rs:214/):9
   3: <wgpu::backend::direct::Context as wgpu::Context>::surface_configure
             at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/wgpu-0.12.0/src/backend/[direct.rs:921](http://direct.rs:921/):13
   4: wgpu::Surface::configure
             at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/wgpu-0.12.0/src/[lib.rs:3267](http://lib.rs:3267/):9
   5: fj_host::graphics::renderer::Renderer::new::{{closure}}
             at ./src/graphics/[renderer.rs:79](http://renderer.rs:79/):9
   6: <core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll
             at /rustc/db9d1b20bba1968c1ec1fc49616d4742c1725b4b/library/core/src/future/[mod.rs:80](http://mod.rs/):19
   7: futures_executor::local_pool::block_on::{{closure}}
             at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/futures-executor-0.3.21/src/local_[pool.rs:315](http://pool.rs:315/):23
   8: futures_executor::local_pool::run_executor::{{closure}}
             at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/futures-executor-0.3.21/src/local_[pool.rs:90](http://pool.rs:90/):37
   9: std::thread::local::LocalKey<T>::try_with
             at /rustc/db9d1b20bba1968c1ec1fc49616d4742c1725b4b/library/std/src/thread/[local.rs:399](http://local.rs:399/):16
  10: std::thread::local::LocalKey<T>::with
             at /rustc/db9d1b20bba1968c1ec1fc49616d4742c1725b4b/library/std/src/thread/[local.rs:375](http://local.rs:375/):9
  11: futures_executor::local_pool::run_executor
             at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/futures-executor-0.3.21/src/local_[pool.rs:86](http://pool.rs:86/):5
  12: futures_executor::local_pool::block_on
             at /home/lukas/.cargo/registry/src/[github.com](http://github.com/)-1ecc6299db9ec823/futures-executor-0.3.21/src/local_[pool.rs:315](http://pool.rs:315/):5
  13: fj_host::main
             at ./src/[main.rs:211](http://main.rs:211/):24
  14: core::ops::function::FnOnce::call_once
             at /rustc/db9d1b20bba1968c1ec1fc49616d4742c1725b4b/library/core/src/ops/[function.rs:227](http://function.rs:227/):5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

It seems to be an issue with webgpu discussed here: https://github.com/gfx-rs/wgpu/issues/2519

ramboldio, 2022-03-09:
changing the power mode in src/graphics/[renderer.rs](http://renderer.rs:44 to
`power_preference: wgpu::PowerPreference::LowPower` did not work.

ramboldio, 2022-03-09:
It worked when switching fully to the intel card, rebooting and then running the project, since then there is no other way than to use the intel graphics.

ramboldio, 2022-03-09:
it feels like, there is not much more to do to catch the case somehow and forward anybody to this issue until wgpu fixes the issue from above? maybe it's only this one card (I have the same as in the issue; that might not be pure coincidence)..

hannobraun, 2022-03-10:
Thank you for opening the issue, @ramboldio!

> It worked when switching fully to the intel card, rebooting and then running the project, since then there is no other way than to use the intel graphics.

Good to know you got it working at least!

> it feels like, there is not much more to do to catch the case somehow and forward anybody to this issue until wgpu fixes the issue from above? maybe it's only this one card (I have the same as in the issue; that might not be pure coincidence)..

That sounds reasonable. However, thinking a bit about it, trying to catch this specific case might be overkill. Maybe we can try to catch *any* panic in wgpu initialization, and add an error message that points to this issue (I think this is the only such issue we currently have). That should be easier.

If we want to be proactive about it, we can open a "graphics hardware bugs meta-issue", where we point to all known such issues, and where people can post new ones.

hannobraun, 2022-03-10:
Labeling as https://github.com/hannobraun/Fornjot/labels/help%20wanted, since I don't have any hardware that causes this failure, and these graphics issues aren't exactly my core expertise (but then again, what is? :grin:).

@ramboldio If you want to try and figure this out, that would be very welcome! Happy to assist in any way I can.

ramboldio, 2022-04-04:
So it's definitley a wgpu problem. Wgpu examples also don't work. The workarounds are: Switch off nvidia card or switch from wayland to X11 or wait for ubuntu 22.04 and hope that they resolve the wayland+nvidia issues by then. I would say, there is not much Fornjot can/should do..

closed 2022-04-04 by ramboldio

hannobraun, 2022-04-05:
Thanks for reporting back, @ramboldio! Hopefully this will resolve itself soon.

jimblandy, 2022-04-18:
From what I am told, it seems that the Vulkan API on NVIDIA is missing functionality (Linux kernel DMA-BUF API support) essential for use with Wayland, and until this is fixed, there is nothing Vulkan users can do.

I assumed that, if `vkcube` (the Vulkan demo program from https://github.com/KhronosGroup/Vulkan-Tools) can display its cube - which it certainly does - then whatever it's doing, `wgpu` should be able to do the same. But I'm told that vkcube is actually failing to report the error that occurs when it tries to use the NVIDIA adapter, and is silently falling back to some other GPU.

hannobraun, 2022-04-19:
Thank you for the info, @jimblandy!

I guess we could try to work around it by detecting the combination of NVIDIA+Wayland and just not use Vulkan in that case. But I worry that such workarounds will become counterproductive, if not properly maintained.

Probably it's best to do nothing and look into higher-level engines/renderers that could shield us from these kinds of problems (by applying said workarounds).