hannobraun/fornjot #430: Face edges not always present in triangulation

hannobraun, 2022-04-05:
It's possible that edges of a face don't show up in the triangulation, which is obviously an invalid result. It's relatively easy to come up with an example, once you know what to look for. Check out this quick sketch I did:

[image]

There's a face on the left, and a possible Delaunay triangulation of that face on the right. The long vertical edge that I specially marked is not present in the triangulation.

Thanks to @alexoro412, who [reported this issue](https://github.com/hannobraun/Fornjot/issues/105#issuecomment-1081050929) in #105!

This should be pretty easy to fix, hopefully. Spade can do [constrained Delaunay triangulations](https://docs.rs/spade/2.0.0/spade/struct.ConstrainedDelaunayTriangulation.html), which should take care of this issue.

Blocked on #105, since I believe we should have a test suite in place before fixing more triangulation issues.

hannobraun, 2022-04-11:
#105 has been addressed. This issue is no longer blocked.

I forgot to label this as https://github.com/hannobraun/Fornjot/labels/status%3A%20blocked in the first place.

hannobraun, 2022-05-04:
I suspect that the missing triangles here are an instance of this bug:
[image]

This is basically the normal star model, except that the `let radius = ...` line has been replaced with this:
``` rust
let mut radius = if i % 2 == 0 { r1 } else { r2 };
radius *= (i + 1) as f64 / num_vertices as f64;
```

willhansen, 2022-07-16:
Here's a more minimal example:
```
#[fj::model]
pub fn model(
) -> fj::Shape {
    let good_x = 0.5;
    let bad_x = 0.4;
    let x = bad_x;
    let mut other = Vec::new();
        other.push([0., 0.]);
        other.push([x, 0.]);
        other.push([0.4, 1.0]);
        other.push([0.1, 0.1]);
        other.push([0., 0.8]);

    let other = fj::Sketch::from_points(other);

    other.into()
}
```
[image]
[image]

hannobraun, 2022-07-16:
Thank you, @willhansen!

willhansen, 2022-07-17:
Converted example to failing unit test: 
https://github.com/hannobraun/Fornjot/compare/main...willhansen:Fornjot:bugfix/%23430-triangulation-missing-edges

hannobraun, 2022-07-18:
That's great, @willhansen, thank you!

Would you mind slapping an `#[ignore]` on the test and submitting that in a pull request? That way, the test is available to be used right where it will be needed, without making the CI build fail.

PR #1369 (2022-11-18): Fix triangulation of sharp, concave faces
Here's a test model that was producing an invalid mesh before:
[image]

Now it works:
[image]

Close #430
closed 2022-11-18 by hannobraun