hannobraun/fornjot #478: CI build should fail, if crates are missing `![deny(missing_docs]`

hannobraun, 2022-04-14:
As of #475 crates are organized into different directories. All crates that are targeted at ecosystem or end users are in [`crates/`](https://github.com/hannobraun/Fornjot/tree/main/crates), internal tools are in [`tools/`](https://github.com/hannobraun/Fornjot/tree/main/tools), and as before, all models are still in [`models/`](https://github.com/hannobraun/Fornjot/tree/main/models).

All user-facing crates should be fully documented. To prevent mistakes, the CI build should check that all crates in the `crates/` directory have a `![deny(missing_docs)]` in their `lib.rs`.

This could be added to the [CI configuration](https://github.com/hannobraun/Fornjot/blob/main/.github/workflows/ci.yml) as an additional job. That job can run only on Linux, like the *Format* and *Clippy Check* jobs. If this isn't easy to do only with GitHub Actions, we can add a new Rust program to `tools/`, and call that from the job using `actions-rs/cargo` (other jobs are already doing this).

Labeling as https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, since this doesn't require lots of knowledge of Fornjot, and could serve as an introduction to the project.

Related issue: #477

freylint, 2022-04-17:
I'm someone who tinkered with porting parts of openscad to rust and am currently considering picking this up as a first issue.

The requirements only force full documentation of user facing crates time of PR. This gives us some ambiguity in how it can be implemented. As I'm new to this project, I'd love for some developer insight into my thoughts on my solutions to this issue.

#### A lint only mechanism
We could try a pass of cargo clippy with all lints but `missing_docs` disabled. This would cause the build to fail on CI without failing on the developer machine.

I've seen a lot of contention on denying this lint, as it can make hacking on the codebase frustrating for some.

#### Parsing Cargo.toml and crawling the crates [^1]
We could alse parse the cargo.toml and check for the lint until some set of code that would forbid top level lint declarations. We could then force every crate in the repo to apply the `missing_docs` lint for every stage of development and CI.

#### A Compromise [^2]
I suggest using the _parse and crawl_ approach to instead enforce a warn of `missing_docs` to avoid workflow disruption and use the _lint only_ mechanism at CI time to fail any PRs which haven't followed the warning.

[^1]: a crawler like this should be integrated with related CI improvements in #477
[^2]: #269 suggests that this is an approach which the maintainers don't disagree with.

hannobraun, 2022-04-17:
Thank you for your analysis, @freylint! I had just added `#![deny(missing_docs)] because that fits my own needs, and didn't consider that this could be frustrating for contributors. So it's good to get another perspective.

I like your suggestion! Warning on missing docs, then failing at CI time if that warning is not heeded, indeed sounds like a good compromise. Although I think we can simplify this and just fail CI on *any* warnings. Having code with warnings is generally not a good idea, I believe, and as a maintainer it can sometimes be easy to miss that a PR introduces them.

> Parsing Cargo.toml and crawling the crates

Just to be clear and avoid misunderstandings, you mean parsing `lib.rs`, right? That's were lints are typically configured.

freylint, 2022-04-17:
> I like your suggestion! Warning on missing docs, then failing at CI time if that warning
> is not heeded, indeed sounds like a good compromise. Although I think we can
> simplify this and just fail CI [...]
I see this pattern of denying all warnings used quite a lot and support adding it to this project.

> > Parsing Cargo.toml and crawling the crates
> 
> Just to be clear and avoid misunderstandings, you mean parsing `lib.rs`, right? That's were lints are typically configured.

I mean parsing the root level Cargo.toml that specifies workspace members. We could then walk the workspace members checking that the lint is enabled in lib/main.rs.

Writing this out had made me realize that #477 should probably be addressed first.

hannobraun, 2022-04-17:
> I see this pattern of denying all warnings (in CI) used quite a lot and support adding it to this project.

:+1: 

> I mean parsing the root level Cargo.toml that specifies workspace members. We could then walk the workspace members checking that the lint is enabled in their lib/main.rs.

Ah, got it. My assumption is that all crates should be workspace members, so whether you scan `Cargo.toml` or just walk the directories, that's just an implementation detail. I don't really care one way or another.

I assume the better approach will reveal itself while doing the work.

> Writing this out has made me realize that #477 should probably be addressed first.

They are definitely related, and I assume that both issues would be addressed would be addressed by the same tool. I'll leave it to you to decide which ones you want to work on, and in which order :smile:

freylint, 2022-05-11:
> I just figured out that the approach I suggested here originally won't work: Only libraries need API documentation, so a tool that's going to complain that `fj-app` doesn't have `#![deny(missing_docs)]` isn't valuable.

I have some concerns about this. Binaries often have helper functions, consts, etc. inside them that should be documented.
I think the issue is false positives, but that might better be approached with an #[allow(...)] annotation. Assuming these false positives are rare enough. I don't personally mind these annotations, but I've seen people expressing annoyance with them as code clutter. Ultimately this is an engineering decision with trade offs.

> 1. Close this issue. It would be nice to have, but not much more. Certainly not critical.

I agree. Solving this would probably require more development time than it's worth. I joined the project to investigate making a UI for it to bring it more in line with OpenSCAD and your communications in the past seem to indicate that that's something you're looking for.
At the very least it did serve as a good first issue! :laughing: 

> 2. Split the crates/ directory into libs/ and apps/, and scan for missing #![deny(missing_docs)] (or, probably better, !#[warn(missing_docs)]) in libs/ only.

If we were to go down this route, we could just add match on if the crate has a main.rs, lib.rs, or both and apply the appropriate lints that way if you don't want to restructure the project.

hannobraun, 2022-05-12:
> I have some concerns about this. Binaries often have helper functions, consts, etc. inside them that should be documented.

I'm not saying that nothing in there should be documented. I'm just saying, it's not public, user-facing documentation, so a `!#[deny(missing_docs)]` (or even a `!#[warn(missing_docs)]`) is probably not appropriate.

> I agree. Solving this would probably require more development time than it's worth.

Yeah, let's close this then.

Thanks for participating here! We didn't bring this to the expected conclusion, but I also won't complain about one less issue to worry about :smile:

closed 2022-05-12 by hannobraun