hannobraun/fornjot #479: `release-operator` should automatically determine which crates to publish

hannobraun, 2022-04-14:
We have (or more accurately, @hendrikmaus has) been working on automating the release process, which is being tracked in #104. [`release-operator`](https://github.com/hannobraun/Fornjot/tree/main/tools/release-operator) is an internal tool that takes care of this.

As of this writing, `release-operator` is getting a list of crates to publish as a command-line argument. The actual list is specified in the CD configuration:
https://github.com/hannobraun/Fornjot/blob/63a7518c29956c9be2bd49b36c868b41f73b2150/.github/workflows/cd.yml#L138-L148

We determined that this is good enough for a start, but I already [made my first mistake](https://github.com/hannobraun/Fornjot/pull/473). It would be much nicer, if `release-operator` could determine by itself, which crates it should publish. As of #475, all user-facing crates are located in the [`crates/`](https://github.com/hannobraun/Fornjot/tree/main/crates) directory, so this should be relatively straight-forward to do.

One complication is, that these crates depend on each other, so the order in which they are published is important. `release-operator` would have to build a dependency graph by analyzing each crate's `Cargo.toml`, to determine a valid order. I'm sure there are libraries that can help with this, but I haven't checked.

Labeling as https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, since this is a project with limited scope, and a broad understanding of Fornjot is not required.

hendrikmaus, 2023-02-27:
I wonder if one could re-use or adapt the code that Cargo uses to render the module tree when invoking “cargo tree”.

hannobraun, 2023-02-27:
Maybe. On the other hand, once we have the list of dependencies for each crate (which we should be able to get with [`cargo_metadata`](https://crates.io/crates/cargo_metadata)), it's probably pretty simple to write an algorithm that orders the crates. I mean, given the input size (~10 crates?), performance is not really a concern. So writing a highly specific `O(n²)` algorithm would be totally fine.

PR #1900 (2023-06-20): Determine crates to publish automatically
I've tested this locally as much as feasible, by inserting debug output. I think it should work in practice, but we'll have to see how to next release goes.

Close #479
closed 2023-06-20 by hannobraun