hannobraun/fornjot #566: Bounding volume of sweeps is incorrect

hannobraun, 2022-05-11:
Sweeps have recently been made more flexible. Now it's possible to sweep along arbitrary straight paths, not just along the z-axis. This seems to have introduced a bug in the AABB calculation.

[image]

As you can see here, the near plane slices through the model, showing us the inside. This shouldn't happen at this camera distance.

I'm pretty sure that this piece of code needs to be updated:
https://github.com/hannobraun/Fornjot/blob/3d5500e95621ebe5c786a886a11eaaaab04bd618/crates/fj-operations/src/sweep.rs#L24-L27

Including the target point to extend the original bounding box is not sufficient. I think we instead need to created a transformed copy (moved by `self.path`) of the bounding box, then merge both bounding boxes.

Labeling https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this most likely just requires a focused change to the code I quoted above.

willhansen, 2022-05-24:
Started out on this a bit.  

Headed in the right direction?

https://github.com/hannobraun/Fornjot/compare/main...willhansen:sweep-fix

gabsi26, 2022-05-24:
My PR is sort of a hack that works for the example provided it might be redundant once you are there @willhansen

hannobraun, 2022-05-24:
Thank you both for looking into this, @willhansen and @gabsi26!

I actually think that @gabsi26's pull request (#623) is spot-on. Maybe I'm missing something, but I believe that it's general enough to support the currently supported capabilities (i.e. sweep along a straight line; of course we'll need to expand it, once curved paths are supported, but thinking about that now is just a case of the perfect being the enemy of the good).

@willhansen, I think your solution is more complicated than it needs to be, and it starts going off the rails a bit in that last commit. If I understand correctly, you orient the 2D shape that is swept, to match the orientation of the path, but that's not actually what the sweep operation is doing. If you sweep a sketch that is parallel to the x-y plane, along a path that is at a non-90° angle to that plane, then that won't rotate the sketch. The original sketch (now the bottom of the sweep) and the top of the sweep will both still be parallel to the x-y plane.

But I do appreciate the unit test you're adding. I haven't reviewed it in detail, but I'd love to merge more testing, if you want to submit it. I also see there's a commit in there that adjusts `.gitignore`. That kind of thing is also always welcome.

PR #623 (2022-05-24): Fix bounding volume issue for swept shapes
This addresses #566

This solution is probably less than ideal,
since the bounding volume gets split up into its vertices,
which then get individually translated.

edit (by @hannobraun):
Close #566 (to make sure the issue is closed automatically when this is merged)
closed 2022-05-24 by hannobraun