hannobraun/fornjot #656: CI build should check, whether `Cargo.lock` is up-to-date

hannobraun, 2022-06-02:
Various changes, like adding/removing/updating dependencies, or adding/removing/updates crates in the workspace, results in automatic updates to `Cargo.lock`. It is possible to then accidentally not commit these updates alongside the change that caused them. If such a partial commit is merged into `main`, the next person to run `cargo build`/`cargo test`/... on `main`, will have their local copy of `Cargo.lock` updated.

This can be inconvenient, confusing, and can get in the way of working with Git history (like running `git bisect`). It would be great, if the CI build would fail, if any of it resulted in changes to `Cargo.lock`. I don't know, if there's an easy way to achieve this. If not, adding a custom tool to `tools/` and calling that from one of the CI jobs (maybe *Clippy*) would be an option.

jeevcat, 2022-07-06:
Could you not just check for any unstaged changes in the repo (or specifically to Cargo.lock) after running `cargo build`?

hannobraun, 2022-07-06:
Yeah, I think this would work!

PR #782 (2022-07-06): Verify that Cargo.lock contains all changes it should
Closes #656
closed 2022-07-06 by hannobraun