hannobraun/fornjot #805: Experiment with alternative model definition syntax

hannobraun, 2022-07-12:
We now have a procedural macro used for defining models, which hides the gnarly details of how host-model interaction works under the hood. Here's what that looks like, as of this writing:

``` rust
#[fj::model]
pub fn model(
    #[param(default = 5, min = 3)] num_points: u64,
    #[param(default = 1.0, min = 1.0)] r1: f64,
    #[param(default = 2.0, min = 2.0)] r2: f64,
    #[param(default = 1.0)] h: f64,
) -> fj::Shape {
    // ...
}
```

The current syntax (or to be precise, a variation thereof) has been proposed by @mdaffin and implemented by @gabsi26. Thank you, both!

*Note:* The current syntax allows each `param` attribute to refer to other parameters, for example to set min/max values. That's neat, but it's mostly a side effect of how the current implementation works, and I'm not sure how easy it would be to support this feature going forward (see #804 for plans that might make that difficult). While it would be nice to have that capability long-term, I think we can ignore it for the purpose of short-term experimentation.

The new syntax has been discussed in #72. There have been more syntax suggestions in that issue, that I think might warrant some experimentation. I'm presenting only the one that I like best, and that I think might be an improvement over the current syntax. Please check out #72 for more, or feel free to post your own favorite syntax in the comments here.

Okay, here's the potential alternative syntax:

``` rust
#[fj::model]
pub fn model(num_points: NumPoints, r1: InnerRadius, r2: OuterRadius, h: Height) -> fj::Shape {
    // ...
}

#[derive(fj::Param(default = 5, min = 3, max = u64::MAX))]
pub struct NumPoints(u64);

#[derive(fj::Param(default = 1.0, min = 0.1))]
pub struct InnerRadius(f64);

// etc.
```

This `fj::Param` trait would be implemented for all common types, so the following would also work:

``` rust
#[fj::model]
pub fn model(num_points: u64, r1: f64, r2: f64, h: f64) -> fj::Shape {
    // ...
}
```

I see two advantages of this syntax over the current one:

1. It spreads the syntactic load a bit. Instead of having one huge chunk of attributes on the model function, you have them on each parameter type.
2. It makes the parameter definitions reusable between functions. This is not relevant right now, but it would be useful, once models can define different types of artifacts.

I think argument 2 is the stronger one here. Imagine a model that defines a 3D model, but also the sketch that the 3D model is based on, for export to a 2D format like DXF. The sketch function would be exposed to the host, and need to deal with model parameters, but it would also be called internally, by the 3D model function. It's not unlikely that you'd want mostly identical parameters between both functions. In such a case, this alternative syntax would prevent redundancy in the definitions.

---

The scope of this issue is to implement this new syntax to see how well it works, and if there are unforeseen hurdles. If successful, this experiment could be merged directly. The implementation of the syntax is in the [`fj-proc`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj-proc) crate, and is re-exported by the [`fj`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj) crate.

This issue is not blocked, but it might be a good idea to complete #804 first, as any new syntax code would need to be updated anyway when that is implemented.

mdaffin, 2022-07-12:
> This fj::Param trait would be implemented for all common types, so the following would also work:

I can see a issues with this. There is no good default for all the different values you can enter. `0` is the logical default value but even in your example it makes no sense for any of the parameter:

```
num_points: u64, r1: f64, r2: f64, h: f64
```

Very likely none of these want to be 0 or even have a minimum of 0. It seems like having these default implementation would encourage people to not specify any bounds even when there should be some.

hannobraun, 2022-07-12:
Hmm, true. On the other hand, it might be annoying having to define a parameter type, if you just want to just try something quickly.

Not sure. Doing it either way has disadvantages in different situations.

hannobraun, 2023-05-15:
This issue is no longer applicable. The affected code [has been removed](https://github.com/hannobraun/fornjot/pull/1822). See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

closed 2023-05-15 by hannobraun