hannobraun/fornjot #807: wgpu warning and errors on startup

hannobraun, 2022-07-13:
When I start `fj-app`, here's what I'm seeing:

```
  2022-07-13T10:50:45.250120Z  WARN wgpu_core::device: Requested size 800x600 is outside of the supported range: Extent3d { width: 3840, height: 2022, depth_or_array_layers: 1 }..=Extent3d { width: 3840, height: 2022, depth_or_array_layers: 1 }
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-core-0.12.2/src/device/mod.rs:4666

  2022-07-13T10:50:45.250448Z ERROR wgpu_hal::vulkan::instance: VALIDATION [VUID-VkSwapchainCreateInfoKHR-imageExtent-01274 (0x7cd0911d)]
        Validation Error: [ VUID-VkSwapchainCreateInfoKHR-imageExtent-01274 ] Object 0: handle = 0x55d90f12a140, type = VK_OBJECT_TYPE_DEVICE; | MessageID = 0x7cd0911d | vkCreateSwapchainKHR() called with imageExtent = (800,600), which is outside the bounds returned by vkGetPhysicalDeviceSurfaceCapabilitiesKHR(): currentExtent = (3840,2022), minImageExtent = (3840,2022), maxImageExtent = (3840,2022). The Vulkan spec states: imageExtent must be between minImageExtent and maxImageExtent, inclusive, where minImageExtent and maxImageExtent are members of the VkSurfaceCapabilitiesKHR structure returned by vkGetPhysicalDeviceSurfaceCapabilitiesKHR for the surface (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-VkSwapchainCreateInfoKHR-imageExtent-01274)
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.12.5/src/vulkan/instance.rs:48

  2022-07-13T10:50:45.250499Z ERROR wgpu_hal::vulkan::instance:         objects: (type: DEVICE, hndl: 0x55d90f12a140, name: ?)
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.12.5/src/vulkan/instance.rs:111
```

This is on an AMD Ryzen 7 5700G with Radeon Graphics (is what `lshw` says).

I don't notice any observable problem related to that, except maybe a noticeable delay on startup (I will open a separate issue for that). It would still be good to look into that and figure out what's going on.

If you *don't* see this error on your setup, please let us know by commenting here!

hannobraun, 2022-07-13:
Oh, forgot to mention: I've been seeing this since egui support was merged (#763).

jeevcat, 2022-07-14:
I see something slightly different on Windows 11:

```
  2022-07-14T10:05:50.861464Z  WARN wgpu_hal::vulkan::instance: Unable to find layer: VK_LAYER_KHRONOS_validation
    at C:\Users\Sam\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-hal-0.12.5\src\vulkan\instance.rs:527

    Finished dev [unoptimized + debuginfo] target(s) in 0.28s
  2022-07-14T10:05:52.068758Z  WARN wgpu_core::device: Requested size 1424x714 is outside of the supported range: Extent3d { width: 1920, height: 1009, depth_or_array_layers: 1 }..=Extent3d { width: 1920, height: 1009, depth_or_array_layers: 1 }
    at C:\Users\Sam\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-core-0.12.2\src\device\mod.rs:4666

  2022-07-14T10:05:52.076503Z  WARN wgpu_core::device: Requested size 800x600 is outside of the supported range: Extent3d { width: 1920, height: 1009, depth_or_array_layers: 1 }..=Extent3d { width: 1920, height: 1009, depth_or_array_layers: 1 }
    at C:\Users\Sam\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-core-0.12.2\src\device\mod.rs:4666
```

hannobraun, 2022-07-14:
> Unable to find layer: VK_LAYER_KHRONOS_validation

I think that means that you haven't installed the Vulkan SDK. At least it was like that for me, on Linux. Reference: https://github.com/hannobraun/Fornjot/issues/157#issuecomment-1077466905

Maybe that's also the reason you're not seeing the validation error I'm seeing?

The error message is definitely not ideal. It's not relevant to end users, and not informative for developers. For the end users, we should consider suppressing wgpu warnings in release mode. Not sure about the developers; maybe we can somehow detect the presence of that warning and print some extra information.

> Requested size 1424x714 is outside of the supported range

Those look similar to my first warning. I'm guessing they have the same cause.

jeevcat, 2022-07-14:
> I think that means that you haven't installed the Vulkan SDK. At least it was like that for me, on Linux. Reference: https://github.com/hannobraun/Fornjot/issues/157#issuecomment-1077466905

You're right. Installing the Vulkan SDK helped. Now the errors I see are very similar to yours, plus an additional warning regarding the present mode.

This is the same PC but booting Linux. I have a NVIDIA GTX 1070.

```
2022-07-14T10:55:06.338926Z  WARN wgpu_core::device: Requested size 800x600 is outside of the supported range: Extent3d { width: 1920, height: 1011, depth_or_array_layers: 1 }..=Extent3d { width: 1920, height: 1011, depth_or_array_layers: 1 }
    at /home/sam/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-core-0.12.2/src/device/mod.rs:4666

  2022-07-14T10:55:06.338992Z  WARN wgpu_core::device: Surface does not support present mode: Mailbox, falling back to FIFO
    at /home/sam/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-core-0.12.2/src/device/mod.rs:4674

  2022-07-14T10:55:06.339636Z ERROR wgpu_hal::vulkan::instance: VALIDATION [VUID-VkSwapchainCreateInfoKHR-imageExtent-01274 (0x7cd0911d)]
	Validation Error: [ VUID-VkSwapchainCreateInfoKHR-imageExtent-01274 ] Object 0: handle = 0x560296d11650, type = VK_OBJECT_TYPE_DEVICE; | MessageID = 0x7cd0911d | vkCreateSwapchainKHR() called with imageExtent = (800,600), which is outside the bounds returned by vkGetPhysicalDeviceSurfaceCapabilitiesKHR(): currentExtent = (1920,1011), minImageExtent = (1920,1011), maxImageExtent = (1920,1011). The Vulkan spec states: imageExtent must be between minImageExtent and maxImageExtent, inclusive, where minImageExtent and maxImageExtent are members of the VkSurfaceCapabilitiesKHR structure returned by vkGetPhysicalDeviceSurfaceCapabilitiesKHR for the surface (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-VkSwapchainCreateInfoKHR-imageExtent-01274)
    at /home/sam/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.12.5/src/vulkan/instance.rs:48

  2022-07-14T10:55:06.339708Z ERROR wgpu_hal::vulkan::instance: 	objects: (type: DEVICE, hndl: 0x560296d11650, name: ?)
    at /home/sam/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.12.5/src/vulkan/instance.rs:111
```

hannobraun, 2022-07-14:
> Surface does not support present mode: Mailbox, falling back to FIFO

I think this is something relatively normal, depending on your hardware. Maybe we can adapt our default though, if FIFO is the more available one.

hannobraun, 2022-07-14:
Now that I think about it, maybe now is not the best time to address these issues. We're not on the latest wgpu release right now, as we're waiting for an updated egui integration to be released.

hannobraun, 2022-07-14:
> Wow, hahaha!! Good memory!!

Wrong! Shitty memory, you were just the most interesting guy I talked to at that conference :smile: 

And maybe I'm misremembering, but I think we talked quite a bit over those days.

> Yes, maybe defaulting to `FIFO` might be the way to go, but not really knowing what these are for `wgpu`, I'm not sure what `Mailbox`-capable users might be giving up; for all I know the two modes could even be mutually exclusive... :)

It's the present mode, if I recall correctly, which governs how specifically rendered pixels are moved to the screen. Now that I think about it, there have been changes to that in the latest wgpu release. The documentation has also improved a lot: https://docs.rs/wgpu/latest/wgpu/enum.PresentMode.html

Probably best to wait until we can upgrade the wgpu version, then choose the new `AutoVsync` mode.

> I hope all is well with you!

Working full-time on my own open source project for very little money. Living the dream! :joy: 

> have recently started to build the electric car I may have mentioned when we met in Berlin (converting my garage to a workspace). Happy to chat about it and find out more about what you are up to if you like. I am `U007D#6955` on Discord if you'd like to chat and get caught up.

Absolutely! To be honest, I always regretted not following up after conference. Nice to be back in touch! (I don't know who I am on Discord (more of a Matrix guy), but I'm sure I can figure out how to contact you.)

hannobraun, 2022-10-13:
I've been looking into the wgpu errors I'm seeing.

---

First, these ones:
```
  2022-10-13T10:05:46.926684Z ERROR wgpu_hal::vulkan::instance: VALIDATION [UNASSIGNED-CoreValidation-Shader-InconsistentSpirv (0x6bbb14)]
        Validation Error: [ UNASSIGNED-CoreValidation-Shader-InconsistentSpirv ] Object 0: handle = 0x55656c1b6c50, type = VK_OBJECT_TYPE_DEVICE; | MessageID = 0x6bbb14 | SPIR-V module not valid: [VUID-StandaloneSpirv-Flat-06202] OpEntryPoint interfaces variable must not be vertex execution model with an input storage class for Entry Point id 107.
  %a_color = OpVariable %_ptr_Input_uint Input

    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.13.2/src/vulkan/instance.rs:48

  2022-10-13T10:23:44.525509Z ERROR wgpu_hal::vulkan::instance:         objects: (type: DEVICE, hndl: 0x560b9658fc50, name: ?)
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.13.2/src/vulkan/instance.rs:111
```

That seems to be due to a Naga bug (https://github.com/gfx-rs/naga/issues/2036), which is already fixed. Should take care of itself once we upgrade to the latest wgpu version (#976).

The second one seems to be a follow-on error. It comes from the same call to the the same callback function.

---

Second, this one:

```
  2022-10-13T10:23:44.527262Z ERROR egui_winit::clipboard: Cannot initialize smithay clipboard without a display handle!
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/egui-winit-0.19.0/src/clipboard.rs:139
```

Since the Clipboard has already been causing trouble (https://github.com/hannobraun/Fornjot/issues/815#issuecomment-1187553111), I've decided to just disable the clipboard entirely. We can revisit this later, preferably once we have a working WASM build. Also see https://github.com/emilk/egui/issues/2138.

---

Finally, these ones:

```
  2022-10-13T10:57:01.908473Z  WARN wgpu_core::device: Requested size 800x600 is outside of the supported range: Extent3d { width: 3840, height: 2022, depth_or_array_layers: 1 }..=Extent3d { width: 3840, height: 2022, depth_or_array_layers: 1 }
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-core-0.13.2/src/device/mod.rs:4974

  2022-10-13T10:57:01.908712Z ERROR wgpu_hal::vulkan::instance: VALIDATION [VUID-VkSwapchainCreateInfoKHR-imageExtent-01274 (0x7cd0911d)]
        Validation Error: [ VUID-VkSwapchainCreateInfoKHR-imageExtent-01274 ] Object 0: handle = 0x55621efb8fb0, type = VK_OBJECT_TYPE_DEVICE; | MessageID = 0x7cd0911d | vkCreateSwapchainKHR() called with imageExtent = (800,600), which is outside the bounds returned by vkGetPhysicalDeviceSurfaceCapabilitiesKHR(): currentExtent = (3840,2022), minImageExtent = (3840,2022), maxImageExtent = (3840,2022). The Vulkan spec states: imageExtent must be between minImageExtent and maxImageExtent, inclusive, where minImageExtent and maxImageExtent are members of the VkSurfaceCapabilitiesKHR structure returned by vkGetPhysicalDeviceSurfaceCapabilitiesKHR for the surface (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-VkSwapchainCreateInfoKHR-imageExtent-01274)
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.13.2/src/vulkan/instance.rs:48

  2022-10-13T10:57:01.908751Z ERROR wgpu_hal::vulkan::instance:         objects: (type: DEVICE, hndl: 0x55621efb8fb0, name: ?)
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-hal-0.13.2/src/vulkan/instance.rs:111
```

For some reason, the window gets a spurious resize event on initialization, which then leads to these errors. I get another resize event to the correct size right after, so this can be taken care of by only handling resizes once per frame. Also see https://github.com/rust-windowing/winit/issues/2094.

---

I've fixed all the warnings/errors that can currently be fixed in #1216.

Labeling as blocked on #976.

PR #1444 (2022-12-12): Update all dependencies to latest versions
This includes `wgpu`, `wgpu_glyph`, `raw-window-handle`, `egui`, `egui-wgpu`, and `egui-winit`.

Close #807 
Close #976
closed 2022-12-12 by hannobraun