hannobraun/fornjot #808: Noticeable delay on startup

hannobraun, 2022-07-13:
Since egui support was merged (#763), I'm seeing a noticeable delay on startup. The window opens immediately, but stays transparent for some time, maybe half a second. After that, the UI and graphics are shown like normal, and I notice no other problems.

This is on an AMD Ryzen 7 5700G with Radeon Graphics (is what `lshw` says). Maybe it's related to #807.

jeevcat, 2022-07-14:
The main delay at startup that I'm seeing is waiting for `cargo build` of the model to finish. Would be nice if this was somehow non-blocking.

hannobraun, 2022-08-30:
I haven't seen any exceptional delays in a while, except when the model took longer to compile for some reason. That might have been the source of the delay in the first place (as @jeevcat noted), or maybe I was seeing something weird that has since cleared up (we have updated relevant dependencies multiple times since then).

I've also opened #1015 to track the specific problem of the GUI not showing before the model is loaded. Closing in favor of that.

closed 2022-08-30 by hannobraun