hannobraun/fornjot #821: Exit with error, if user provides unused parameters

hannobraun, 2022-07-15:
Fornjot models can have parameters. The user can pass those to the app via the command-line interface, using `--parameters`.

If the user passes any parameters that the model doesn't use, maybe because of a typo or because the model has changed since the user last used it, that parameter is ignored. This is quite error-prone.

If the user passes a parameter that the model doesn't use, this should cause the app to exit with an appropriate error message. Detecting this requires a better interface between host and model. See #804, which this issue is currently blocked on.

Michael-F-Bryan, 2022-07-23:
The metadata system mentioned in https://github.com/hannobraun/Fornjot/issues/804#issuecomment-1193139276 would be quite useful here.

hannobraun, 2022-08-12:
#804 has been addressed. This is no longer blocked.

hannobraun, 2023-05-15:
This issue is no longer applicable. The affected code [has been removed](https://github.com/hannobraun/fornjot/pull/1822). See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

closed 2023-05-15 by hannobraun