hannobraun/fornjot #829: Zoom direction is reverse of standard

willhansen, 2022-07-16:
"Scroll-up" is usually zoom in.

eg: google maps

Throwing in some negatives here should fix:
[image]

hannobraun, 2022-07-16:
This is intentional. The metaphor used is that you manipulate the model, not an abstract camera. So in terms of zoom, that means you pull it closer or push it away.

And *standard* is a difficult word here, because "mouse wheel forward" no longer means "scroll up" in all cases. As far as I can tell, this change to the previous standard was inspired by the rise of touch interfaces, where you move the content, not some abstract viewport. Makes sense to do it the same way with a mouse (although I have to admit, this specific case is one I didn't bother getting used to, and changed it in the configuration).

So yeah, there's no clear right and wrong here, and I decided that having a consistent metaphor is more important than following the prevailing mouse wheel zoom convention. Whether that was the right decision is up for discussion, and I'm happy to have that discussion here.

For the time being, I'm going to close this issue as "working as intended" though.

willhansen, 2022-07-16:
After looking into it a bit more, apparently the "scroll up to zoom out" convention is more common in CAD programs than I realized.

hannobraun, 2022-07-16:
This issue inspired to test again, whether I can get used to "natural scrolling" (as it's called in the Gnome settings). Turns out that changing this setting not only inverts scrolling, but *all* mouse wheel behavior. Including Google Maps and (drum roll) Fornjot.

This is definitely not intended. As I said before, the intention was to provide a consistent metaphor, not invert whatever metaphor is used at the operating system level.

Not sure what the right thing is here. First reaction:
- Invert default scrolling behavior, so it follows operating system settings. It stands to reason that the kind of person that enables "natural scrolling" might be more inclined to Fornjot's current default behavior than the majority of the population.
- Provide a setting to allow people to invert Fornjot's scrolling behavior, if their expectation is not met.

An alternative would be to try and detect the operating system behavior and use that to always provide a consistent UI experience. But that seems like more trouble than it's worth.

In any case, I'm reopening this issue, and am going to classify it as a bug. Suggestions are welcome.

hannobraun, 2022-07-16:
Oh, and if people could try how that works on other platforms and report back (if your platform allows you to configure mouse wheel behavior), that would be very appreciated!

PR #1204 (2022-10-11): Invert default zoom direction; add config to override that
From one of the commit messages:

> The intent behind the original zoom direction was to provide a consistent metaphor: You manipulate the model, not an abstract camera.
>
> Unfortunately, it turned out to not be that simple. At least on Gnome, if you enable "natural scrolling" (which is the same metaphor, applied to the Gnome UI), you invert all mouse whell input, and so also the zoom direction in Fornjot.
>
> With this change, the zoom direction in Fornjot follows the "natural scrolling" configuration in Gnome. Unfortunately I don't  [... cut]
closed 2022-10-11 by hannobraun