hannobraun/fornjot #830: Window always starts maximized

willhansen, 2022-07-16:
But should it?

[image]

hannobraun, 2022-07-16:
This is not some deep decision; it's just the way I preferred it so far during development. I think as a default, it's probably as good as any.

It seems that many applications remember the window size (but not necessarily location) from the last time they were started. That's probably the right thing for Fornjot too. What do you think?

ObiWanRohan, 2022-12-17:
Is the window size and location persistance an OS feature or an application feature?

hannobraun, 2022-12-17:
I honestly don't know. It's certainly not something that just happens, according to my experience at least. So *if* it is an OS/platform feature, at the very least it is something you have to opt into somehow.

hannobraun, 2023-05-15:
Closing as out of scope. See [A New Direction](https://www.fornjot.app/blog/a-new-direction/) for context.

While the affected code is still part of the repository, it only serves as a development tool now. I think for that purpose, the current behavior works perfectly.