hannobraun/fornjot #849: `#[fj::model]` should check signature of model function

hannobraun, 2022-07-20:
The `#[fj::model]` proc macro provides a convenient way for users to define Fornjot models. The attribute is applied to a function, and isolates the user from the low-level implementation details of the host/model interface.

Model functions need to return an `fj::Shape`, but right now, this compiles:

``` rust
#[fj::model]
pub fn model() {
    todo!()
}
```

There is still a compiler error, if the wrong type is used. This fails to compile:

``` rust
#[fj::model]
pub fn model() {
    "invalid return value"
}
```

And this is the error:

```
error[E0308]: mismatched types
 --> models/error/src/lib.rs:3:5
  |
1 | #[fj::model]
  | ------------ expected `Shape` because of return type
2 | pub fn model() {
3 |     "invalid return value"
  |     ^^^^^^^^^^^^^^^^^^^^^^ expected enum `Shape`, found `&str`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `error` due to previous error
```

Michael-F-Bryan, 2022-07-23:
One strategy I've used several times is to keep the original function unchanged and assign it to a variable so you can ensure it has the right signature. This has the nice side-effect of leaving the original function accessible of other crates want to import it (i.e. so the `test` model doesn't need to copy/paste star and friends into `lib.rs`).

For example, this is how users [register a plugin](https://github.com/Michael-F-Bryan/fornjot-plugins/blob/7ced70b260fe173f3fd08bb3b3ebb3ca6614981b/crates/plugins/src/abi/mod.rs#L36-L58) in my plugins experiment:

```rust
#[macro_export]
macro_rules! register_plugin {
    ($init:expr) => {
        #[no_mangle]
        #[doc(hidden)]
        pub fn fornjot_plugin_init(
            host: &mut dyn $crate::Host,
        ) -> Result<$crate::PluginMetadata, $crate::Error> {
            // Note: explicitly require a particular function signature.
            let init: fn(&mut dyn $crate::Host) -> Result<$crate::PluginMetadata, $crate::Error> =
                $init;

            init(host)
        }
    };
}
```

It gets a bit more tricky with `#[fj::model]` though because we're playing around with the parameter list, and there's no way to leave a `fn`'s parameter list unspecified (i.e. `let model: fn(...) -> fj::Shape = model` is a syntax error).

hannobraun, 2022-07-24:
Thanks for the insight, @Michael-F-Bryan!

> It gets a bit more tricky with `#[fj::model]` though because we're playing around with the parameter list, and there's no way to leave a `fn`'s parameter list unspecified (i.e. `let model: fn(...) -> fj::Shape = model` is a syntax error).

The proc macro has access to the parameter list though, so re-creating it in the `let model: fn` shouldn't be too much of a problem, I think.

PR #867 (2022-07-24): Add typechecking to model signatures
This fixes #849 by placing the original model function inside our generated `extern "C" fn model()`, letting the Rust compiler make sure it type checks. From there, it's simple to call the original function with its parsed arguments and assert that the result is convertible to a `fj::Shape`.

I've also added some examples to the attribute's docs so we can make sure `#[model] fn model() { todo!() }` doesn't compile.
closed 2022-07-24 by hannobraun