hannobraun/fornjot #850: Implement `fj-app --version`

hannobraun, 2022-07-20:
`fj-app` should have a `--version` argument that returns the current version. Given that we release weekly, it would be great if this version wouldn't have to be updated manually. This should be straight-forward to achieve: https://doc.rust-lang.org/cargo/reference/environment-variables.html#environment-variables-cargo-sets-for-crates

While I would definitely merge a pull request that implements this straight-forward version, it would be great if there was some additional nuance:

- The ID of the Git commit that the binary was built from should be output as well.
- It should be clear whether the binary is from a published release or not.

I assume that both of those can be achieved by creating the version string in a build script, and passing it to the `fj-app` source code as an environment variable. The release automation could use an environment variable to signal the build script whether this is an official release build.

closed 2022-07-28 by hannobraun

hendrikmaus, 2022-07-28:
Clap has the long version for this and a convenience Macro to help setting it during compile time https://docs.rs/clap/latest/clap/builder/struct.App.html#method.long_version

To figure out if we are building a release or not, release-operator's deduce step must run before. 

I would propose to add a job that runs before the binary build job. It only runs the deduce command and shares the outcome as an output with the remainder of the pipeline. Then the binary builder job can wait for that and use the output. 

And after building the binaries, the remaining release-operator job also runs conditionally on the output from the first job that ran the deduce command. 

I don't think you need a build.rs or anything more than Clap and a bit of pipeline step reordering.

hannobraun, 2022-07-28:
On second thought, we might want to use the version in other parts of the UI, not just as a response to `--version`. We might even want to have it available in other applications in the future, that don't rely on Clap. Or maybe we'll have multiple CLI applications that all want to display the same version.

In any case, it might be a disadvantage to rely on Clap for this capability. Unless the current `build.rs`-based solution causes any maintenance issues, I think it makes sense to stay with it.

PR #868 (2022-07-28): Embed the git commit in fj-app's version number
Fixes #850.

The version number for an official build will look something like this:

```console
$ FJ_OFFICIAL_RELEASE=1 cargo run --quiet -- --version
fj-app 0.8.0 (66de87d0)
```

Not having the `$FJ_OFFICIAL_RELEASE` variable set will add `unreleased` to the version number, and having a dirty git tree will add a `-modified` suffix to the commit hash.

```console
$ cargo run --quiet -- --version
fj-app 0.8.0 (4ef5109e-modified, unreleased)
```

When `git` doesn't start (i.e. because it's not installed), we silently leave out the commit hash. Thi [... cut]