hannobraun/fornjot #856: Add UI element that displays status updates

hannobraun, 2022-07-21:
Right now, a lot of what's happening with the Fornjot app is only visible on the command-line. This is not user-friendly, since the Fornjot app opens a window, and the user might not expect to receive important information anywhere else. Depending on how the app is started, the user might not even have access to the command-line output.

There are certainly many ways to display status updates in a user-friendly way, but I'd like to start with a single UI element that is flexible enough to display all kinds of status updates: some kind of scrollable text field (I'm sure [`egui`](https://crates.io/crates/egui) provides a suitable widget).

Off the top of my head, here's what status updates I think should go into that text field:

- Reloading model (due to a change to the model source code)
- Model reloaded
- Error reloading model (with full error message)

Since this information comes from places like [`fj-host`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj-host), which doesn't have any concept of UI, it probably makes sense to have some kind of struct with methods like `report_error`, `report_status_update` (just suggestions; not to be taken too literally) that just stores the message as a string, and can then be used to display the message where appropriate.

In the regular UI use case, that would mean [`fj-viewer`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj-viewer) queries that struct for messages and displays them in the aforementioned text field. When running [`fj-app`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj-app) on the command-line with the `--export` argument, which would not result in a window being opened, `fj-app` could use the same struct to display the messages on the command-line.

I think the appropriate place for that struct would be in [`fj-interop`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj-interop).

devanlooches, 2022-08-02:
Is it alright if I take a stab at this?

hannobraun, 2022-08-02:
Of course, go ahead! Let me know, if you have any questions.

Thank you for your interest!

devanlooches, 2022-08-03:
Hey, Is there an easy way to force the window (if possible only the config part) to redraw in order to reflect the newest status update?

hannobraun, 2022-08-03:
That shouldn't be necessary. The window is redrawn every frame, so hopefully 60 times for second (or slower, depending on your hardware). It shouldn't be necessary to force a redraw, because the next one should be happening in a few milliseconds anyway.

egui is an immediate-mode GUI, meaning there isn't any state in the UI components. The whole UI is rebuilt every frame, so it shouldn't be necessary to do anything special to have it reflect the latest status. Please note that I haven't done our egui integration, nor have I worked with this code so far. I might be missing some subtleties.

What specifically is your problem? Maybe there's something else going on that isn't related to redrawing?

devanlooches, 2022-08-03:
Hmm in that case it might be my struct isn't being updated for some reason? For some reason only the first message is being drawn in the ui.

hannobraun, 2022-08-03:
Sounds reasonable. If you want, you can open a pull request and mark it as a draft, and I'll have a look.

devanlooches, 2022-08-03:
Sure I will do that shortly

devanlooches, 2022-08-03:
I opened #911

hannobraun, 2022-08-12:
This has been addressed in #911 and #919. Thank you, @devanlooches! Closing.

closed 2022-08-12 by hannobraun