hannobraun/fornjot #920: ``just build`` fails on export-validator step

devanlooches, 2022-08-05:
Error:

```
cargo run -p export-validator
   Compiling remove_dir_all v0.5.3
   Compiling fastrand v1.8.0
   Compiling tempfile v3.3.0
   Compiling export-validator v0.1.0 (/Users/devan/Workspace/Fornjot/tools/export-validator)
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-m64" "-arch" "x86_64" "/var/folders/gk/_t6yht456mv08x6lpqygypx00000gn/T/rustcTiBDiV/symbols.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1231r3rw062mmh2d.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.12fjpzo0tfptdjx2.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.12fy4ilem562je78.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.12zjmyxu9olfljru.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.13jsjii9g9jyvlsv.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.154zky3afxmxbfh1.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.185r24yajoyqjkwo.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.187apyhwwjpd67rh.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1c9zscoqvjfjhzy8.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1dihppg9slfs0y2y.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1dqyhcjhfrrvfwv.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1fab21p06q157jew.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1fsllcvb9tox8cka.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1goxfy5wis8ishhm.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.1k05g471nbb2x5gq.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.2d27ifh3xpdzv0iz.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.2d6ljp3yg8bjwfio.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.2gr8694jqwh53zyi.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.2ko4vla39o8gdcw0.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.2q9f9ytqvqx3ld05.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.2vkxsztuwdd9prj0.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.368jdgp9o0wz5msu.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.38zu7526609pcha.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3asn5nvg91mfg6o2.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3jzd4m5icdguqgl0.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3ko6szsls2f25ekm.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3ndfy4qe6y02f62m.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3t2glgbkeo8r2nkg.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3t75to8c8k0i1kpz.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3zyf42dnevcf2yf1.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.41i7dm8n796pphey.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.48frywl45i6lqv5f.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.48mu6v5o8oscxjlm.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.48tyadli8xjretdu.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4am2ysoyqrh2xsov.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4e1hzafif90r9pbl.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4e623szoz6y74j2h.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4eqzbfu3zf2vc2j5.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4fzeh2yl7t8onpnm.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4g2nijpv6m133zs5.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4gxwus2bjtf5x164.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4hn76v3jf6zxxaqj.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4jhhbh36c2npywbo.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4kj1vqi7ly6gx9bz.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4llvt873nvpkzlgu.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4mlo5n8wcfzqgl4h.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4oxzsxo94hvlcjh3.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4q2zuqn2dt95hwv2.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4s85le6xr8fhzwo9.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.4tbyiodh3wg2neb3.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.508k4v6nig6xo532.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.52dm076d64oti6bv.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.53u0klh9v3ohzhow.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.59s02wv2knnmpkhs.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.5b385hhvbv0f68y2.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.5f68yovt30g4pcl3.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.8x795sxxcmyamb7.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.byq9wo9xo3wp89l.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.d5oua6qgl55qesm.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.ge9o22slvci8gjd.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.ita385w0we2iyd5.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.jnz4r6o7v61limz.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.tb1hw3cvr8lkg6j.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.vapckrrqx2kjn49.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.vc4nvqbzjpsui0a.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.zp40bzsi7f127gx.rcgu.o" "/Users/devan/Workspace/Fornjot/target/debug/deps/export_validator-419be965c839a003.3jj6nluqb0ujzaij.rcgu.o" "-L" "/Users/devan/Workspace/Fornjot/target/debug/deps" "-L" "/Users/devan/Workspace/Fornjot/tools/export-validator/lib3mf/libs" "-L" "/Users/devan/.rustup/toolchains/1.62.1-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib" "-l3mf" "/Users/devan/Workspace/Fornjot/target/debug/deps/libtempfile-dc917f1e56097537.rlib" "/Users/devan/Workspace/Fornjot/target/debug/deps/liblibc-6a29003711e39f17.rlib" "/Users/devan/Workspace/Fornjot/target/debug/deps/libcfg_if-e6fce2b47f1e3a40.rlib" "/Users/devan/Workspace/Fornjot/target/debug/deps/libfastrand-f74c0b19f076aa35.rlib" "/Users/devan/Workspace/Fornjot/target/debug/deps/libremove_dir_all-18d8f41122dc5eca.rlib" "/Users/devan/Workspace/Fornjot/target/debug/deps/libanyhow-d9580bbe45cb675a.rlib" "/Users/devan/.rustup/toolchains/1.62.1-x86_64-apple-darwin/lib/rustli [... cut ...]
                export_validator::validate_model::h547745ef40494711 in export_validator-419be965c839a003.4fzeh2yl7t8onpnm.rcgu.o
            "_lib3mf_object_isvalid", referenced from:
                export_validator::validate_model::h547745ef40494711 in export_validator-419be965c839a003.4fzeh2yl7t8onpnm.rcgu.o
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)

error: could not compile `export-validator` due to previous error
error: Recipe `build` failed on line 8 with exit code 101
```

hannobraun, 2022-08-06:
Thank you, @devanlooches! What platform is that? Looks like macOS with Intel hardware?

`export-validator` relies of `lib3mf`, and I've added pre-compiled lib3mf libraries for the repository to support that. I just had a look, and it seems I only added the files for Linux, which is the only platform I can test.

To support `export-validator` on other platforms, we need the pre-compiled libraries for those too. Here: https://github.com/hannobraun/Fornjot/tree/main/tools/export-validator/lib3mf

That directory has a README that says where the files come from. Basically, we need to add the right files from the SDK, maybe necessary symlinks to make sure the linker finds them, and then possibly adapt `export-validator`'s [`build.rs`](https://github.com/hannobraun/Fornjot/blob/main/tools/export-validator/build.rs) to make it work (maybe not, I don't know).

Any help doing that would be very appreciated! Labeling https://github.com/hannobraun/Fornjot/labels/help%20wanted, as I don't have access to the missing platforms. Also labeling https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this is a build/linker issue that doesn't require any knowledge of Fornjot to figure out.

devanlooches, 2022-08-06:
Sorry for not giving more info. You are correct I'm on a macOS with Intel hardware. I tried putting the files in the libs/ directory but I'm still getting the same error message.

hannobraun, 2022-08-08:
> Sorry for not giving more info. You are correct I'm on a macOS with Intel hardware. I tried putting the files in the libs/ directory but I'm still getting the same error message.

For Linux, I had to make sure that the file has a very specific name, hence the symbolic link there. This required some experimentation though, and I'm a bit out of my depth there, especially regarding macOS.

hannobraun, 2022-09-07:
I just noticed @Michael-F-Bryan's [lib3mf bindings](https://github.com/Michael-F-Bryan/lib3mf). It seems it falls [back to compiling lib3mf from source](https://github.com/Michael-F-Bryan/lib3mf/blob/d655cc3ddb9a1cffb518eb719cd9d365fbb61e48/sys/build.rs#L9-L12), if it can't find compiled binaries. Maybe using it could be the solution to this issue?

Michael-F-Bryan, 2022-09-08:
My original intention behind creating that `lib3mf` crate was actually so I could publish it to crates.io and switch Fornjot's validator over to it.

Unfortunately, I couldn't get Windows building reliably so I had to [disable it in CI](https://github.com/Michael-F-Bryan/lib3mf/commit/aeef47f431791a75ef77e33a48a5024f8a7e5f0c)... Skimming through [these build logs](https://github.com/Michael-F-Bryan/lib3mf/runs/7743917166?check_suite_focus=true), I suspect it's an issue with their code and build system not being compatible with MSVC++. However, I hate debugging build systems for C/C++ projects so when I saw those errors and wasn't able to fix it easily, I threw the project in the too-hard basket and moved onto something else :disappointed: 

It works on MacOS, though!

hannobraun, 2022-09-08:
Thanks for the info, @Michael-F-Bryan! Let's see how this shakes out.

I also didn't want to mess around with any C/C++ builds, which is why I decided to just check the library files into Git. Not a bad approach actually, I think.

DengFX2010105421, 2022-10-12:
from /Fornjot/tools/export-validator/lib3mf/README.md,  just download  [lib3mf_sdk_v2.2.0.zip](https://github.com/3MFConsortium/lib3mf/releases/download/v2.2.0/lib3mf_sdk_v2.2.0.zip),unzip it,

hannobraun, 2022-10-12:
Thank you for looking into this, @DengFX2010105421! I'd appreciate a pull request. I'd prefer not to do it myself, since I can't test it.

hannobraun, 2022-10-17:
I've come to the conclusion that it's a good idea to run the Export Validator for all platforms on the CI build, not just Linux. I tried to implement this today, using GitHub Actions as my test environment.

It was a huge pain, but I managed to do so for macOS here: https://github.com/hannobraun/Fornjot/pull/1226

I absolutely failed to get it working for Windows, and I don't understand what the problem is. It just won't work. Here's my work-in-progress: https://github.com/hannobraun/Fornjot/pull/1230

Any help figuring this out would be highly appreciated!

hannobraun, 2022-10-18:
Thanks, @Michael-F-Bryan, but I'd rather not. I've spent way too much time trying to link yesterday, and I have no intention to spend more time trying to compile :smile: 

What I'll do is, I'll replace this issue with a more specific one that summarizes the current situation and what's wrong, and try to get some help dealing with it. I hope there will be enough interest from Windows users to take care of this. If not, I can live without `export-validator` supporting Windows, for now.

hannobraun, 2022-10-18:
Closing in favor of #1231.

closed 2022-10-18 by hannobraun