hannobraun/fornjot #943: 2D boolean operations

hannobraun, 2022-08-11:
It should be possible to combine sketches (2D shapes) using union, difference, and intersection operations. This is related to 3D boolean operations (#42, #43, #44), which I'm currently working on. While the low-level details are going to be different (and most likely much simpler for 2D), the high-level algorithm should be pretty much the same, allowing for reuse there.

I'm not sure where this issue fits into the current plans. It could make sense to finish 3D boolean operations first, then work on making the algorithm more generic, until it supports 2D operations too. It could also make sense to go the other way: implement 2D boolean operations first (which should be much easier, given the simpler geometry), then work to expand the algorithm to also support 3D.

hannobraun, 2022-09-01:
It's always a trade-off! On the one hand, what you say is totally correct. On the other hand, getting something working (and be useful) earlier would be worth a lot, as it could provide experience and feedback, which could be worth the extra work generalizing the 2D code to 3D.

I think if I were to start fresh, then I would do 2D first, then later generalize to 3D. I have no intent to do that though, and since I've already put a lot of work into the 3D part, I will continue along that path.

joamatab, 2023-01-03:
We already have Boolean operations in gdsfactory, I hope you find it useful

hannobraun, 2023-01-03:
I hadn't hear of gdsfactory. Looks interesting!

closed 2023-11-30 by hannobraun