hannobraun/fornjot #985: failed to load model

AndyJado, 2022-08-24:
## what I did

0. both `wsl` & `macos13`; both build from `source` & `crate.io`
1. `pwd` and I am under `models/`
2. `models % ` :     `fj-app --model cuboid`

## Error:

```
Error: Failed to load model: models/cuboid

Caused by:
    0: I/O error while loading model
    1: No such file or directory (os error 2)
```

hannobraun, 2022-08-24:
Hey @AndyJado, thank you for opening this issue!

I assume this is the `models/` directory in this repository? If so, `fj-app` might be picking up [the configuration file](https://github.com/hannobraun/Fornjot/blob/main/fj.toml), which defines `models/` as `default_path`, which might cause `fj-app` to try and load `{$repository_root}/models/models/cuboid`. (We're using [Figment](https://crates.io/crates/figment) for configuration, and I'm actually not sure if that picks up configuration files from parent directories, so this might be wrong.)

You could try the following things:

- Run `fj-app` from the repository root.
- Delete `fj.toml`, try again from the `models/` directory.

If I'm right, then this isn't a bug. But there is definitely a lot of room for improvement:

- The error message needs to be **much** better. It should show the absolute path of what it tried to load. It should display the configuration, to explain *why* it tried to load that path.
- The current behavior is a bit questionable. Maybe, if you pass a relative path to `--model` that points to a valid model, that path should always take precedence over any configuration.

AndyJado, 2022-08-24:
yeah it works now.

closed 2022-08-24 by AndyJado

hannobraun, 2022-08-24:
Good to hear that, @AndyJado! I've opened #986 and #987 to track improvements that could reduce the confusion here.