hannobraun/fornjot #986: Improve error message, if model can't be loaded

hannobraun, 2022-08-24:
When starting [`fj-app`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj-app), you tell it which model to load by passing a command-line argument (`-m`/`--model`). If that model can't be found, this results in an error message. This error message can be hard to understand, and lacks information that could help to make sense of the problem.

Here's the error message that this currently results in:
```
Error: Failed to load model: models/cuboid

Caused by:
    0: I/O error while loading model
    1: No such file or directory (os error 2)
```

There should be a lot more information here, to help the user determine *why* the model couldn't be loaded:

- In addition to the relative path shown here, also show the absolute path, in case that makes the error more obvious.
- Explain, step-by-step, how the path came to be:
  1. The current working directory.
  2. The [default path from the configuration](https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-app/src/config.rs#L12).
  3. The model passed via `-m`/`--model`, or the [default model from the configuration](https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-app/src/config.rs#L13) as a fallback.

This information should help users recognize what went wrong.

The code that builds the path and creates the error message is located in `fj-app`:
https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-app/src/main.rs#L49-L59

I'm labeling this as https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this is a fairly isolated change that doesn't require a lot of knowledge about the context.

ArshErgon, 2022-08-24:
@hannobraun do you mean something like this?
```
Error: Failed to load model: models/cuboid
path: usr/Desktop/Fornjot/default_path/src/random.js
Caused by:
    0: I/O error while loading model 'random.js'
    1: Could not find "random.js" file inside 'src'
```

ArshErgon, 2022-08-24:
@hannobraun  do you mean something like this?

```
Error: Failed to load model: models/cuboid
path: usr/Desktop/default_path/model/file.js
Caused by:
    0: I/O error while loading model 'file.js'
    1: Could not find 'file.js' inside 'model', file not present

```

hannobraun, 2022-08-24:
Hey, @ArshErgon!

The "Caused by" part and everything below is provided by [Anyhow](https://crates.io/crates/anyhow), so I don't think we can change much there. We have full control over the error message on top though.

Maybe something like this:

```
Error: Failed to load model: /home/hanno/projects/fornjot/models/models/cuboid
1. Working directory is `/home/hanno/projects/fornjot/models/`
2. Default model path is `models/` (defined in `/home/hanno/projects/fornjot/fj.toml`)
3. Looking for models in combination of 1. and 2.: `/home/hanno/projects/fornjot/models/models`
4. Model provided is `cuboid` (defined in command-line arguments)
5. Can't find directory `cuboid` in `/home/hanno/projects/fornjot/models/models`

Caused by:
    0: I/O error while loading model
    1: No such file or directory (os error 2)
```

Just an idea. There are certainly many ways to present this.

hannobraun, 2022-08-24:
> and how can I run the project on ubuntu? I download rust should I follow this https://github.com/hannobraun/Fornjot/blob/main/CONTRIBUTING.md

Sorry, I missed this question initially.

How to run the project is [explained in `README.md`](https://github.com/hannobraun/Fornjot/blob/main/README.md#usage). [`CONTRIBUTING.md`](https://github.com/hannobraun/Fornjot/blob/main/CONTRIBUTING.md) has some useful info about contributing to the project. Certainly worth a read, but it doesn't explain how to run it.

I think it would be a good idea to add more explanation to `CONTRIBUTING.md` though. I've made a note about this so I don't forget, but until I can get to it, here's what info seems to be missing:
- You can run `fj-app` from the repository root using just `cargo run`. This will load the [test model](https://github.com/hannobraun/Fornjot/tree/main/models/test) by default.
- You can load a different model by passing the model name, for example `cargo run -- -m cuboid`.
- There's a configuration file, [`fj.toml`](https://github.com/hannobraun/Fornjot/blob/main/fj.toml), that makes it so you don't have to pass `-m models/cuboid` (for example) and can just pass `-m cuboid` instead.

hannobraun, 2022-08-25:
I've added the information I outlined in my previous comment to [`CONTRIBUTING.md`](https://github.com/hannobraun/Fornjot/blob/main/CONTRIBUTING.md) (#995). The CI build is still running, but this should get merged in a few minutes.

ArshErgon, 2022-08-25:
@hannobraun yeah, the project is running, while playing with the current settings I found when you click the `Show egui settings UI` inside there's `painting` inside there's a scroll drag `feathering` when I tried to increased it all the checkboxes and those scroll buttons and the panel blur out, like layer blur. is it a issue?

hannobraun, 2022-08-25:
> while playing with the current settings I found when you click the `Show egui settings UI` inside there's `painting` inside there's a scroll drag `feathering` when I tried to increased it all the checkboxes and those scroll buttons and the panel blur out, like layer blur. is it a issue?

I don't know. Probably not?

I wasn't the one who added the egui GUI, and I'm still not very familiar with egui. My understanding is, that the egui settings where left in as a debugging tool. It will be removed or hidden sooner or later. Since you can play with settings in there that a normal user shouldn't really have access to, it's not a surprise that you can get weird effect.

That said, I don't know what "feathering" means, and thus I don't know what this is supposed to do. I have no reason to believe that it's an issue, but I can't know for sure.

PR #1154 (2022-10-13): Error handling
for the issue #986 I have re-edited the error the Fornjot application.
Do review the PR

edit by @hannobraun:
Close #986 (to close the issue when this is merged)
closed 2022-10-13 by hannobraun