hannobraun/fornjot #987: Only use default path, if `--model` doesn't actually point to a model

hannobraun, 2022-08-24:
When starting [`fj-app`](https://github.com/hannobraun/Fornjot/tree/main/crates/fj-app), you tell it which model to load by passing a command-line argument (`-m`/`--model`). The argument passed isn't always interpreted as a relative path though: There is [configuration](https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-app/src/config.rs#L11-L14) that can affect from where the model is loaded, namely the `default_path` key. See the [`fj.toml`](https://github.com/hannobraun/Fornjot/blob/main/fj.toml) from this repository for an example.

This can be confusing, as the user might not be aware that a configuration file exists, and that it affects the loading behavior. I think the following behavior would be clearer:

1. If the argument passed via `-m`/`--model` is a path, either absolute, or relative to the current working directory, where a Fornjot model is found, load that model.
2. Only if no model is found, fall back to using `default_path`.

The code responsible for the current behavior lives in `fj-app`:
https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-app/src/main.rs#L49-L59

Some thoughts on the implementation:

- I can think of two ways to determine whether a directory contains a model:
  1. Do a check for `Cargo.toml` (maybe in [`fj_host::Model::from_path`](https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-host/src/lib.rs#L53-L76)) and return an error if it doesn't exist.
  2. Try to load the model at that path normally, and retry with the fallback path, if there's an error that indicates the model doesn't exist. This method could be more reliable, but would likely require a rework of [`fj_host::Error`](https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-host/src/lib.rs#L380-L433) to make it possible/easy to determine that a model doesn't exist there, as opposed to an error occurring with an existing model.
- The [API of `fj_host::Model`](https://github.com/hannobraun/Fornjot/blob/67cc32db7cf4b757413ebebf075c8330c90fa723/crates/fj-host/src/lib.rs#L40-L252) is a bit weird, with the two load methods and the separate constructor. It seems to be that way because that's convenient for the implementation, not because it's a sensible API for the call sites. This is certainly not necessary, but maybe this issue is an opportunity to rethink that and clean it up a bit.
- It *could* make sense to move some of the logic for determining from where to load a model into `fj-host`. However, `fj-host` shouldn't need to worry about whether something came from command-line argument or the configuration file. These are concepts of `fj-app`.

Labeling as https://github.com/hannobraun/Fornjot/labels/good%20first%20issue. Even though the issue provides some opportunity for redesigning APIs, it is very limited in scope, making it suitable for anyone not experienced with Fornjot's code base yet.

Also see #986, which is related

whereistejas, 2022-09-11:
Hey @hannobraun!
I was thinking about working on refactoring the `Model` methods. I have sketched out what I would like the interface to look like... Would love to hear your thoughts...

Current codebase:
```rust
impl Model {
    pub fn from_path(path: PathBuf) -> Result<Self, Error> {}
    pub fn load_once(
        &self,
        arguments: &Parameters,
        status: &mut StatusReport,
    ) -> Result<fj::Shape, Error> {}
    pub fn load_and_watch(
        self,
        parameters: Parameters,
    ) -> Result<Watcher, Error> {}
}

// crates/fj-host/src/lib.rs
fn main() {
    ...

    let model = Model::from_path(path.clone())
        .with_context(|| format!("Failed to load model: {}", path.display()))?;
    let parameters = args.parameters.unwrap_or_else(Parameters::empty);

    let shape_processor = ShapeProcessor {
        tolerance: args.tolerance,
    };

    if let Some(path) = args.export {
        let shape = model.load_once(&parameters, &mut status)?;
        let shape = shape_processor.process(&shape)?;

        export(&shape.mesh, &path)?;

        return Ok(());
    }

    let watcher = model.load_and_watch(parameters)?;

    run(watcher, shape_processor, status)?;
    Ok(())
}
```

What I would like it to be like:
```rust
impl Model {
    pub fn load_file(path, arguments, status) -> Result<Watcher, Error> {
        from_path();
        load_once();
        load_and_watch();
    }

    pub fn load_file_with_export(path, arguments, status) -> Result<(Watcher, sj::Shape), Error> {}
}

// crates/fj-host/src/lib.rs
fn main() {
    ...
    
    if let Some(path) = args.export {
        let (watcher, shape) = model.load_file_with_export(path, arguments, status);
        let shape = shape_processor.process(&shape)?;
        export(&shape, &path)?;
    } else {
        model.load_file(path, arguments, status)
        
        run(watcher, shape_processor, status)?;
    }

    Ok(())
}
```

hannobraun, 2022-09-12:
Thank you for looking into this, @whereistejas! That proposal looks very reasonable. I'll leave some more comments on the pull request you opened.

hannobraun, 2022-10-26:
I've decided to close this, for the following reasons:
- This would complicate the model path logic, possibly leading to as much confusion as it prevents.
- Specifying a default path might be relatively rare anyway. I just added it because it was convenient in this repository.
- The error message is much better now. See below.

Let's say you ignore the default path from the configuration in this repository and specify the full relative path to the model:
```
cargo run -- --model models/test
```

If you do this, this is what you get:
```
Error: Failed to load model: `models/test`
- Passed via command-line argument
- Path of model: /home/hanno/Projects/main/fornjot/models/models/test
- Searching inside default path from configuration: /home/hanno/Projects/main/fornjot/models

Suggestions:
- Did you mis-type the model path `models/test`?
- Did you mis-type the default path `models`?
- Did you accidentally pick up a local configuration file?

Caused by:
    0: I/O error while loading model
    1: No such file or directory (os error 2)
```

I think that's pretty good!

closed 2022-10-26 by hannobraun