hannobraun/fornjot #1014: egui UI is blurry, mouse input is offset

hannobraun, 2022-08-30:
As the title says, the egui UI on the left is very blurry. The mouse cursor isn't registered where it actually is. See this screenshot:
[image]

The blurriness is a bit subtle on this screenshot (although it's very clear in the original, especially if you've seen the difference). But you can clearly see that the mouse cursor is hovering over "Render original UI", but "Show egui settings UI" is activated.

I'm observing the problem on my Linux/Gnome setup (AMD GPU, although not sure that makes a difference). I'm pretty sure this came in with the recent egui/wgpu/winit upgrade (https://github.com/hannobraun/Fornjot/pull/975). The problem seems to be related to the scale factor of my HiDPI display. If I change the scale factor to 100% in the gnome settings, it works as expected.

I suspect that something has changed in regards to HiDPI handling in either winit or egui. Maybe the solution will become obvious when going over the changelogs.

devanlooches, 2022-09-29:
Just as extra information, I am getting the same result on my Mac.

hannobraun, 2022-09-30:
Thanks for the info, @devanlooches! I think this increases the likelihood of this being a HiDPI display issue. Good to know it's not some platform-specific problem. That would be annoying.

hannobraun, 2022-10-12:
I've looked into this, but no luck. I couldn't find anything that looked suspicious or was obviously missing.

hannobraun, 2022-10-20:
I just discovered that this happens to me with X.org, but not with Wayland.

hannobraun, 2022-10-25:
Fixed in #1266.

Thanks again, @erenoku!

closed 2022-10-25 by hannobraun