hannobraun/fornjot #1143: Dependabot is currently not working

hannobraun, 2022-09-27:
Rust 1.64.0 comes with support for a new Cargo feature: [workspace inheritance](https://doc.rust-lang.org/cargo/reference/workspaces.html#the-package-table). I started using this pretty much immediately (#1132), as it makes the version update that is part of the weekly release much easier, on top of consolidating a lot of redundant metadata.

Unfortunately, Dependabot doesn't support workspace inheritance yet: https://github.com/dependabot/dependabot-core/issues/5315

This means, Dependabot no longer does anything for this repository. It just fails with an error every week. This is unfortunate, but right now I'm not considering rolling the change back. I think the benefits from workspace inheritance outweigh the benefits from Dependabot. I'm opening this issue to document and track the problem. Labeling as https://github.com/hannobraun/Fornjot/labels/status%3A%20blocked.

poliorcetics, 2022-10-05:
I opened a PR for it https://github.com/dependabot/dependabot-core/pull/5794, it should work but it's not merged yet.

It's probably possible to regenerate the docker container and use it locally to make the necessary changes but I have no idea how to do that

hannobraun, 2022-10-05:
Thank you, @poliorcetics, that's awesome! Hope it gets merged soon!

> It's probably possible to regenerate the docker container and use it locally to make the necessary changes but I have no idea how to do that

I have no idea either. I don't know anything about how Dependabot works, and not a lot about Docker either. For now, I'm happy to run `cargo update` manually every week or so.

closed 2022-10-10 by hannobraun