hannobraun/fornjot #1223: Panic when trying to load example model

themmes, 2022-10-14:
First of all, love what you're up to here and hope to be able to contribute at some point!

# Context

Trying to run Fornjot for the first time

```
tomhemmes@Toms-MacBook-Pro dev % fj-app --version
fj-app 0.19.0 (unreleased)
tomhemmes@Toms-MacBook-Pro dev % fj-app --model fornjot/models/spacer
```

I'm running on a 2018 Macbook Pro (Intel) with Big Sur v11.6.8

# Expected

I expected a screen to open and display the `spacer` model.

# Actual

Instead I see the window open for a brief moment and then crash

```
tomhemmes@Toms-MacBook-Pro dev % fj-app --model fornjot/models/spacer 
thread 'main' panicked at 'Error reloading model: LibLoading(DlOpen { desc: "dlopen(/Users/tomhemmes/dev/fornjot/target/debug/libspacer.dylib, 5): image not found" })', /Users/tomhemmes/.cargo/registry/src/github.com-1ecc6299db9ec823/fj-host-0.19.0/src/lib.rs:322:25
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
tomhemmes@Toms-MacBook-Pro dev % RUST_BACKTRACE=1 fj-app --model fornjot/models/spacer
thread 'main' panicked at 'Error reloading model: LibLoading(DlOpen { desc: "dlopen(/Users/tomhemmes/dev/fornjot/target/debug/libspacer.dylib, 5): image not found" })', /Users/tomhemmes/.cargo/registry/src/github.com-1ecc6299db9ec823/fj-host-0.19.0/src/lib.rs:322:25
stack backtrace:
   0: rust_begin_unwind
             at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/panicking.rs:584:5
   1: core::panicking::panic_fmt
             at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/core/src/panicking.rs:142:14
   2: fj_host::Watcher::receive
   3: fj_window::run::run::{{closure}}
   4: <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event
   5: winit::platform_impl::platform::app_state::Handler::handle_nonuser_event
   6: winit::platform_impl::platform::app_state::AppState::launched
   7: winit::platform_impl::platform::app_delegate::did_finish_launching
   8: <unknown>
...
  25: <unknown>
  26: objc::rc::autorelease::autoreleasepool
  27: winit::platform_impl::platform::event_loop::EventLoop<T>::run
  28: winit::event_loop::EventLoop<T>::run
  29: fj_window::run::run
  30: fj_app::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

Could you let me know what I am doing wrong? Please let me know if I need to provide more info or context.

hannobraun, 2022-10-14:
Thank you for the kind words and for opening this issue, @themmes!

First of all, that's a horrible error message, and we should fix that! I'll look into that next week.

Second, I can't reproduce your issue. It looks like you're trying to use an `fj-app` that you compiled yourself (as the version says "unreleased") to load a model from the Fornjot repository, while you're in the repository's parent directory. I tried that and it works for me, so unless I got something wrong, this might be a platform-specific issue.

I don't have access to macOS, and I don't know when someone has last tried that. It looks like our CI build doesn't actually run the app for all platforms (only Linux; another thing I can look into next week), so we might just have broken macOS at some point.

I suggest you try the following:

1. Go to the repository root and do a `cargo run`, or a `cargo run -- -m spacer`. I don't expect a different result, but this could rule out that something really weird is going on with the path the app tries to load the file from.
2. The app fails to load `/Users/tomhemmes/dev/fornjot/target/debug/libspacer.dylib`, so maybe it's getting the file name wrong. Could you check what other files are in that directory?

Thank you!

themmes, 2022-10-14:
I installed `fj-app` using `cargo install fj-app`, which gives me the following:

```
~/dev $ cargo install fj-app
    Updating crates.io index
  Installing fj-app v0.19.0
   Compiling autocfg v1.1.0
   Compiling proc-macro2 v1.0.46
....
   Compiling fj-math v0.19.0
   Compiling fj-interop v0.19.0
   Compiling fj-kernel v0.19.0
   Compiling fj-viewer v0.19.0
   Compiling fj-host v0.19.0
   Compiling fj-export v0.19.0
   Compiling fj-operations v0.19.0
   Compiling fj-window v0.19.0
    Finished release [optimized] target(s) in 2m 46s
  Installing /Users/tomhemmes/.cargo/bin/fj-app
   Installed package `fj-app v0.19.0` (executable `fj-app`)
~/dev $ fj-app --version
fj-app 0.19.0 (unreleased)
```

On your points:

1. Oh, 🎉 using `cargo run -- -m spacer` works perfectly fine! No errors and I can see the model in the application.
2. The contents of `target/debug` look like this

```
~/dev $ fj-app --model fornjot/models/spacer
thread 'main' panicked at 'Error reloading model: LibLoading(DlOpen { desc: "dlopen(/Users/tomhemmes/dev/fornjot/target/debug/libspacer.dylib, 5): image not found" })', /Users/tomhemmes/.cargo/registry/src/github.com-1ecc6299db9ec823/fj-host-0.19.0/src/lib.rs:322:25
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
~/dev $ ls -lah fornjot/target/debug/
total 537k
drwxr-xr-x  10 tomhemmes      staff     320 2022-10-14 18:06 .
drwxr-xr-x   5 tomhemmes      staff     160 2022-10-14 18:06 ..
-rw-r--r--   1 tomhemmes      staff       0 2022-10-14 18:06 .cargo-lock
drwxr-xr-x  15 tomhemmes      staff     480 2022-10-14 18:06 .fingerprint
drwxr-xr-x   8 tomhemmes      staff     256 2022-10-14 18:06 build
drwxr-xr-x 107 tomhemmes      staff    3.3k 2022-10-14 18:06 deps
drwxr-xr-x   2 tomhemmes      staff      64 2022-10-14 18:06 examples
drwxr-xr-x   5 tomhemmes      staff     160 2022-10-14 18:06 incremental
-rw-r--r--   1 tomhemmes      staff    1.3k 2022-10-14 18:06 libspacer.d
-rw-r--r--   1 tomhemmes      staff    531k 2022-10-14 18:06 libspacer.rlib
```

hannobraun, 2022-10-14:
I think I know what it is! Here: https://github.com/hannobraun/Fornjot/pull/1209

Version 0.19 still requires models to specify the `crate-type` in `Cargo.toml`, which is no longer necessary on `main`. And since `crate-type` has been removed from the models in the repository, those are incompatible with the released version right now.

This should clear up on Monday, when I release 0.20. We should address this properly by checking for version mismatch between a model and the app (https://github.com/hannobraun/Fornjot/issues/980).

> I installed `fj-app` using `cargo install fj-app`, which gives me the following:

Oh, interesting. It shouldn't say "unreleased" in the version then. Yet another thing that's broken :smile: 

---

So while the main issue is basically a specific instance of #980 (and therefore a duplicate), I'd like to leave this issue open for the time being, until we either address or open more specific issues for the smaller problems we've uncovered here (bad error message, CI only runs app on Linux, version says "unreleased" when it shouldn't). I'll take care of next week!

hannobraun, 2022-10-19:
Version 0.20 has been released on Monday, and since it includes #1209, it this issue is no longer present.

In addition, I've either fixed (https://github.com/hannobraun/Fornjot/pull/1226, https://github.com/hannobraun/Fornjot/pull/1235) or opened more specific issues for (https://github.com/hannobraun/Fornjot/issues/1231, https://github.com/hannobraun/Fornjot/issues/1236) any of the smaller problems mentioned here.

Closing as duplicate of #980.

closed 2022-10-19 by hannobraun