hannobraun/fornjot #1272: Remove `-m`/`--model`

hannobraun, 2022-10-26:
`fj-app` can be provided with the `-m`/`--model` argument, to specify which model to load. And in fact, this is the only command-line argument that isn't really optional (strictly speaking it is, but only if a default model is specified in the configuration file).

It would be simpler to just do away with `-m`/`--model` and use a positional argument instead. So instead of this:
```
fj-app -m model
```

We'd have this:
```
fj-app model
```

I think this would be more natural. In fact, I forget the `-m`/`--model` all the time.

Labeling https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this only requires surface-level knowledge of Fornjot. This probably requires changes only in [`fj-app`'s `args.rs`](https://github.com/hannobraun/Fornjot/blob/main/crates/fj-app/src/args.rs).

kopackiw, 2022-11-07:
Happy to help on that with my PR above!

Also, big thanks for very helpful documentation both in the app, as well in this issue - this is my first OSS contribution in Rust and it felt really easy to dive in! 🙇🏻

hannobraun, 2022-11-08:
Thanks for taking care of this issue, @kopackiw!

> Also, big thanks for very helpful documentation both in the app, as well in this issue - this is my first OSS contribution in Rust and it felt really easy to dive in! 🙇🏻

Glad to hear that! I do my best to make things easy to get into.

That doesn't always work though (limited resources and all), so if you find some code you don't understand, always feel free to open issues to request clarification.

PR #1323 (2022-11-08): Turn the "--models" CLI option into an argument
This PR removes the `--model` flag in the CLI in favour of an optional `model` argument.
Tested locally on custom inputs & examples from the documentation - no regression found.

Closes #1272
closed 2022-11-08 by hannobraun