hannobraun/fornjot #1280: Print warning, if full version of app and model don't match

hannobraun, 2022-10-27:
Currently, if the version of the Fornjot app doesn't match the version of Fornjot that a model uses, this will result in an error when the model is loaded. This works well enough for released versions, but the it might lead to confusion when in-development versions are involved. This might be the case, if a user tries to load an example model from this repository using a released version of the Fornjot app.

The version is determined at build time in [the `fj` crate's `build.rs`](https://github.com/hannobraun/Fornjot/blob/147f7e8cbe12875c5705296eb4c463c8735f210a/crates/fj/build.rs). From there, it is [included in the `fj` crate](https://github.com/hannobraun/Fornjot/blob/147f7e8cbe12875c5705296eb4c463c8735f210a/crates/fj/src/version.rs#L5-L20), and the `static`s defined there are then [used at model-load-time in `fj-host`](https://github.com/hannobraun/Fornjot/blob/147f7e8cbe12875c5705296eb4c463c8735f210a/crates/fj-host/src/model.rs#L107-L121) to check for a mismatch and potentially generate the error. Two different versions are determined and included in `fj`: The package version (something like `1.2.3`), and a "full" version which includes more information, like the Git commit hash. Only the package version is used for the version check.

It would be too restrictive to use the full version as the basis for the version mismatch check, but since a mismatch there *might* cause mysterious problems, it's probably best to print a warning, explaining the situation. I'm not sure what the ideal solution is, but for a start, it would probably be enough to log a warning-level log message to stderr.

Labeling https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, as this only requires surface-level knowledge of Fornjot, and I've laid out all the moving parts here.

zthompson47, 2022-11-02:
Hello!  I put together a PR for this issue that prints a warning to stderr.

Testing is interesting because there is a new symbol in the shared library that renders all old models incompatible:
```
Error receiving updated shape: Failed to load the Fornjot version that the model uses
- Is your model using the `fj` library? All models must!
- Was your model created with a really old version of Fornjot?

Caused by:
    /home/zach/fornjot/models/shape/target/debug/libshape.so: undefined symbol: version_full
```
To test, I made a temporary clone of the repository with the updated symbol, and made a commit to get a different full version.  I then used that repository in the `Cargo.toml` of the model crate to trigger a full version mismatch.

Is there any need for automated tests here?  I suppose it might be tricky and/or unnecessary (at least for now).

hannobraun, 2022-11-02:
Thank you for looking into this, @zthompson47!

Your approach to testing sounds very reasonable. I don't think we need an automated test for this. If someone came up with a reasonable one, I wouldn't say no, but I think this falls into this quadrant of testing where the test would be hard to write but the benefit would be low. Not worth losing sleep over, in my opinion.

I'll take a look at your pull request now!

PR #1300 (2022-11-02): Warn on full version mismatch of host and model
Fixes #1280
closed 2022-11-02 by hannobraun