hannobraun/fornjot #1307: Model version check is disabled on Windows

hannobraun, 2022-11-04:
When loading a model, the Fornjot app checks the Fornjot version that this model model uses, to prevent mystery errors due to incompatibilities. This check is currently disabled on Windows[^1]. There's a bug that prevents it from working there. See #1303 for context.

It would be great if we fixed that bug, and enabled the same version check on all platforms. Unfortunately, I don't have access to a Windows machine and lack Windows expertise. Labeling as https://github.com/hannobraun/Fornjot/labels/help%20wanted, as help would be especially appreciated!

[^1]: Well, actually it's not. But it will be. I'm opening this issue prematurely, because I'm currently working on disabling it, and I'd like to print a warning that points to this issue.

hannobraun, 2022-11-15:
I went over the version comparison code with a fine comb and found (and fixed!) a potential soundness hole. I don't know if that was actually the problem reported in #1303, but if it was, it's possible that the version check can be enabled on Windows without further changes.

MartinKavik, 2022-11-15:
I've just pulled the latest `main` and deleted the condition for Windows and tried `println!` the versions and everything seems to work as expected now on Windows 10. Good job!  

Btw is it possible to somehow filter `debug!` messages for a specific crate or a project? When I set `RUST_LOG=DEBUG` I see only a sea of `wgpu_*` and `fj_viewer*` messages.

hannobraun, 2022-11-15:
> I've just pulled the latest `main` and deleted the condition for Windows and tried `println!` the versions and everything seems to work as expected now on Windows 10. Good job!

Thank you! Such a subtle thing. I wasn't very optimistic that this would actually solve it!

> Btw is it possible to somehow filter `debug!` messages for a specific crate or a project? When I set `RUST_LOG=DEBUG` I see only a sea of `wgpu_*` and `fj_viewer*` messages.

Yes: `RUST_LOG=fj_host=debug cargo run`

I think you can go even more fine-grained, by specifying this on the module level. And you can probably chain multiple directives using `,`, but I might not remember the syntax correctly.

PR #1374 (2022-11-21): Enable model version check on Windows
From what I've heard it works now, as of https://github.com/hannobraun/Fornjot/pull/1358.

Close #1307
closed 2022-11-21 by hannobraun