hannobraun/fornjot #1499: Fornjot app freezes after a while

hannobraun, 2023-01-11:
After having the Fornjot app open for a bit, while I was working on a model, the app froze:

[image]

On the terminal, it showed the following error:
```
  2023-01-11T09:29:08.279377Z ERROR wgpu::backend::direct: Handling wgpu errors as fatal by default
    at /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-0.14.2/src/backend/direct.rs:2402

thread 'main' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_buffer
      note: label = `egui_index_buffer`
    Buffer size 1181699064 is greater than the maximum buffer size (268435456)

', /home/hanno/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-0.14.2/src/backend/direct.rs:2403:5
```

I don't know what is causing this, but it could be related to #1498, as I've been seeing a bunch of those warnings.

kazatsuyu, 2023-01-11:
I saw a similar error in https://github.com/hannobraun/Fornjot/issues/1424#issuecomment-1347313383. I will try to reproduce it.

hannobraun, 2023-01-12:
I saw the same error just now, while trying to reproduce #1498, with two notable differences:

1. There weren't any of those epaint warnings, so those issues might not be related.
2. The app immediately crashed instead of freezing.

I can't tell what I did to trigger this. I'll keep looking into it.

kazatsuyu, 2023-01-12:
If `wgpu::SurfaceError` occurs, `Renderer::draw()` will return early.
https://github.com/hannobraun/Fornjot/blob/cdaceb97a65e69fd89b94ef3ff14c98c22a0c8c4/crates/fj-viewer/src/graphics/renderer.rs#L250-L263

And `context.end_frame()` in `Gui::prepare_draw()` will not be executed.

https://github.com/hannobraun/Fornjot/blob/cdaceb97a65e69fd89b94ef3ff14c98c22a0c8c4/crates/fj-viewer/src/gui.rs#L283

Probably, not calling `end_frame()` and adding shapes to be drawn continuously is causing the vertex count to exceed the limit.

hannobraun, 2023-01-12:
Thank you, @kazatsuyu! I actually came to a similar conclusion just now.

Your note that `end_frame` needs to be called to prevent this is super-valuable though! I didn't recognize this, and was struggling to come up with a reliable fix.

I'm going out for lunch now, but I'll try moving the `end_frame` into the `update` method, to see if that fixes the issue. Thank you!

hannobraun, 2023-01-12:
Your fix worked, @kazatsuyu! I'm pretty sure I would have needlessly complicated this, if you hadn't helped. Thanks again!

closed 2023-01-12 by hannobraun