hannobraun/fornjot #1507: Don't panic, when creating a polygon from an empty list of points

hannobraun, 2023-01-13:
When passing an empty list of points to [`fj::Sketch::from_points`](https://github.com/hannobraun/Fornjot/blob/55bb16e8198073c14c6eb378a906c3bd8bec0497/crates/fj/src/shape_2d.rs#L112-L118), the application will panic:

```
thread 'host' panicked at 'Attempted to compute a bounding box from an empty sketch', crates/fj-operations/src/sketch.rs:129:17
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread 'main' panicked at 'Host thread panicked', crates/fj-host/src/host.rs:74:25
```

This panic originates from `fj-operations`:
https://github.com/hannobraun/Fornjot/blob/55bb16e8198073c14c6eb378a906c3bd8bec0497/crates/fj-operations/src/sketch.rs#L129-L132

There is also a second source of panic in the same file:
https://github.com/hannobraun/Fornjot/blob/55bb16e8198073c14c6eb378a906c3bd8bec0497/crates/fj-operations/src/sketch.rs#L60-L63

While creating an empty polygon is maybe not something one would do intentionally, it can easily happen while working on a model. After all, the model is live-reloaded as you edit the code. So this definitely should not panic. *Maybe* it should print a warning to the GUI, but I'm not even sure about that.

We should remove this panic, and make sure that everything still works when creating an empty sketch. There might be follow-on errors, I don't know. If during that process it becomes clear that a warning message would be a good idea, it could be added too, but I think it's perfectly fine to just not do it and possibly leave that to follow-up work.

hannobraun, 2023-01-13:
I don't know how hard it will be to address this, as it's unclear if there are any follow-on problems that removing these panics would uncover. But at the very least, removing these panics and reporting back how that went is a very straight-forward task. Labeling as https://github.com/hannobraun/Fornjot/labels/good%20first%20issue for that reason.

tmayoff, 2023-02-01:
Decided to look into this, since I almost immediately encountered this issues the first time I tried it.

Just removing the assert on 129 caused this: 
https://github.com/hannobraun/Fornjot/blob/55bb16e8198073c14c6eb378a906c3bd8bec0497/crates/fj-operations/src/sketch.rs#L136

Returning Aabb::default instead

```rust
let segments = poly_chain.to_segments();
let mut points = vec![];

segments.is_empty() {
    return Aabb::default();
}
```
This caused some more panics here, https://github.com/hannobraun/Fornjot/blob/55bb16e8198073c14c6eb378a906c3bd8bec0497/crates/fj-kernel/src/objects/full/cycle.rs#L31-L34

Then removing that caused this to get hit, https://github.com/hannobraun/Fornjot/blob/55bb16e8198073c14c6eb378a906c3bd8bec0497/crates/fj-kernel/src/objects/full/cycle.rs#L45-L47

hannobraun, 2023-02-02:
Thanks for investigating, @tmayoff!

The root cause isn't easy to fix, unfortunately. This is going to require some non-trivial design changes to the core data structures. Good news is, I already have those planned (sorry, no issue open yet; it's on my task list).

Meanwhile, maybe it would be possible to paper over this closer to where the first panic happens? My idea is, if the sketch is empty, instead of trying to create an empty sketch in the kernel, maybe we can not create a sketch at all? The `Shape` implementation for `fj::Sketch` could [produce an `Option<Sketch>` instead of a `Sketch`](https://github.com/hannobraun/Fornjot/blob/075c0f047c1fceb91d9cb938bbd7193010f64e2c/crates/fj-operations/src/sketch.rs#L19), maybe. This would have follow-on effects on its callers and I'm not sure about the practicality. But it's something we could try.

If that doesn't work out, we'll have to leave this open until the core data structure redesign is done.

hannobraun, 2023-02-10:
Oh yeah, I think that would work! The user can just `unwrap` for now, which would be way better than the current situation, as a panic in the model [no longer crashes the app](https://github.com/hannobraun/Fornjot/pull/1534)!

Maybe add a note to the doc comment of that method, explaining that this is a temporary situation.

Thanks for continuing to look into this, @tmayoff!

hannobraun, 2023-02-20:
Closing, since @tmayoff addressed this in https://github.com/hannobraun/Fornjot/pull/1585 (thanks again!).

I considered keeping this open, as the solution is not the permanent one. Given that the API available to models is very much in flux anyways, and there are a lot of things that could be improved, I don't think it's worth keeping an issue open to track that one `Option`.

closed 2023-02-20 by hannobraun