hannobraun/fornjot #1568: Errors from retrieving model arguements are swallowed

hannobraun, 2023-02-07:
The Fornjot application communicates with the models it loads over an FFI boundary (with Rust code on both sides). This is necessary, because models are compiled to dynamic libraries, and Rust doesn't have a stable ABI. Panics in model code must not cross this FFI boundary, as that would be undefined behavior. Therefore, panics in model code are caught and turned into errors, which are then displayed in the Fornjot application.

There is one hole in the code that is handling this. If a panic occurs in a model's argument handling code (which is generated by `fj-proc`), it is converted into an error, but then later silently ignored:
https://github.com/hannobraun/Fornjot/blob/ed08c4c8d99d3e146a9a815406e02e276a040d77/crates/fj/src/abi/context.rs#L73

This is not ideal, as that could hide bugs in a highly confusing manner. Instead, the method this is happening in (`Context::get_argument`) should return a `Result` instead of an `Option` and pass that error on. Any downstream code should be adapted as necessary.

Labeling as https://github.com/hannobraun/Fornjot/labels/good%20first%20issue, since this is a pretty localized problem.

emcloughlin, 2023-04-01:
I think I will try to tackle this as a first issue. Wish me luck lol

hannobraun, 2023-04-01:
Sounds great! Looking forward to what you come up with!

closed 2023-05-15 by hannobraun